pybamm-team/PyBaMM #3913: Investigate alternatives for `xarray` to handle `ProcessedVariable` computations

agriyakhetarpal, 2024-03-20:
Recently, #3892 highlighted that `pandas` was being installed as an implicit required dependency for PyBaMM, because it was a required dependency for one of our required dependencies (`xarray`). `pandas` was otherwise listed as an optional dependency with the `[pandas]` extra and is currently used only for handling CSV files.

This dependence on `xarray` is particularly concerning, because:

1. If `pandas` decides to act upon [PDEP-10](https://pandas.pydata.org/pdeps/0010-required-pyarrow-dependency.html) with v3, it would drastically increase the download size for PyBaMM (`pyarrow` wheels across platforms are 120+ megabytes in size at a minimum).
2. This would have complications on if things like Pyodide support are considered – where running PyBaMM on the browser would require excess bandwidth utilisation and slow down usage workflows. It would also affect regular users by a bit in Google Colab, where Python virtual environments and dependencies are not saved or cached.

Prior to the use of `xarray` (see #2366) as a backend for the `ProcessedVariable` and the `ProcessedVariableComputed` classes, the [`scipy.interpolate`](https://docs.scipy.org/doc/scipy/reference/interpolate.html) module was being used – which could be an option to return to.

There is time until `pandas` decides on this and also until we release v24.5, so we can take into account some of the developments around this area as they arise.

kratman, 2024-03-20:
What is pyodide being used for if it is an issue?

I have used pyarrow and pandas in a lot of web based apps without issue. Both pandas and pyarrow are pretty common in data science, so I know these get used in web/notebook applications on a regular basis

agriyakhetarpal, 2024-03-20:
> What is pyodide being used for if it is an issue?

It's not being used by us currently, but as a part of my work assignment I am extending support for it across a lot of PyData projects and across the Scientific Python ecosystem (please see https://github.com/Quansight-Labs/czi-scientific-python-mgmt/issues/18 and https://github.com/Quansight-Labs/czi-scientific-python-mgmt/issues/19). PyBaMM isn't quite there yet, because we have CasADi as a dependency—it is tricky to compile it to WASM—if it becomes optional, we could move things forward on that (see #3826). The best and most stable example of where you can see Pyodide currently is on [any of the usage examples in the `scikit-learn` documentation](https://scikit-learn.org/stable/auto_examples/ensemble/plot_gradient_boosting_regression.html#sphx-glr-auto-examples-ensemble-plot-gradient-boosting-regression-py), where you can bring interactive docs via client-side JupyterLite notebooks.

> I have used pyarrow and pandas in a lot of web based apps without issue. Both pandas and pyarrow are pretty common in data science, so I know these get used in web/notebook applications on a regular basis

There's no issue as such if you do so locally for any data science workflows because the `pyarrow` backend is extremely fast, but 1. those with unstable connections can have issues running such notebooks online, and 2. having a heavy (required) dependency graph in general isn't good for any library (packaging/distribution, for example, is one of the areas). But this is a smaller part of the picture; some of the responses on https://github.com/pandas-dev/pandas/issues/54466 are quite insightful in this regard.

agriyakhetarpal, 2024-03-20:
We did have pandas as an optional dependency before #3892, didn't we? I imagine it should not be a lot of work to make it fully optional back again with the `import_optional_dependency` wrapper. Or are we using it in a notebook where we haven't installed it in the introductory code cell?

A lot of the plotting features (for example `matplotlib`) were set as optional so that you were not forced to use it, and therefore you could use libraries like `holoviz`, `pyvista`, `altair`, `seaborn`, or any others of your choice offering a plotting backend and a graphics module. It is still optional at this time but in PyBaMM's history before v23.5 it was one of the "truly" optional dependencies (but we didn't have a list of optional dependencies back then).

agriyakhetarpal, 2025-06-19:
I took a look at it, and it is still relevant. The pandas team will proceed with PyArrow as a required dependency, and having xarray still has the potential to negatively impact many users due to an increased download size. The Pyodide situation is resolved, though – PyArrow has WASM wheels now, and they are not too large (10.9 MiB or so). However, we need progress on https://github.com/apache/arrow/issues/38536 that concerns more conventional, non-WASM environments.

agriyakhetarpal, 2025-06-19:
SciPy is a good alternative still (we do have the newer `RegularGridInterpolator` or the legacy `RectBivariateSpline` to replace the deprecated `interp2d`), and there's also the fact that `pandas` is listed as an optional dependency, while it is not – that needs to be addressed. 

We're essentially using `xarray` for reating labelled arrays with coordinate information, and performing interpolation with the `.interp()` method. We need this kind of pattern here:

We have:

```python
self._xr_data_array = xr.DataArray(entries, coords=[(dim_name, coords), ("t", t_pts)])
return self._xr_data_array.interp(**kwargs).values
```

and we could replace it with something like this:

```python
from scipy.interpolate import RegularGridInterpolator
self._interpolator = RegularGridInterpolator((coords, t_pts), entries, 
                                           method='linear', bounds_error=False, fill_value=np.nan)

# and then in __call__
points = np.column_stack([kwargs[dim] for dim in self.dims])
return self._interpolator(points)
```

Also xref https://github.com/pydata/xarray/issues/3981