pybamm-team/PyBaMM #3925: Flaky test cases.

prady0t, 2024-03-24:
While running unit-tests, some tests appaers to be flaky. See https://github.com/pybamm-team/PyBaMM/pull/3857#issuecomment-2015300918

#3857 Is a prerequisite to this. Once it gets merged, we can discuss more such tests here and come up with a solution for them.

agriyakhetarpal, 2024-03-24:
For additional context – we have seen this happen just with the JAX evaluator tests, where we do not receive consistent outputs.

Right now, a stopgap solution is to run these tests again by adding `pytest-flaky` as a dependency and using a `@pytest.mark.flaky` decorator on them – it will then run those tests thrice until they (hopefully) pass

agriyakhetarpal, 2024-04-02:
Upon a closer look, these failures seem to be coming from comparing the time taken to get to a second solution after the first solution, which speaks more about how the test has been written rather than the computation, I feel. It can be inherently flaky, especially with how we have been noticing that the these tests in parallel mode are quite slow, cf. #3948.

PR #3963 (2024-04-04): Remove time check from Jax tests to improve stability
# Description

Jax tests checked runtime, which is inconsistent in the CI suite. This should keep the same coverage, but stop checking the runtime. Note: the changelog is also updated since that was missed in #3961 

Fixes #3925

## Type of change

Only test code.

- [x] Bug fix (non-breaking change which fixes an issue)

# Key checklist:

- [x] No style issues: `$ pre-commit run` (or `$ nox -s pre-commit`) (see [CONTRIBUTING.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CONTRIBUTING.md#installing-and-using-pre-commit) for how to s [... cut]
closed 2024-04-04 by kratman