pybamm-team/PyBaMM #3935: [Bug]: Error reported when using the casadi solver

YoungAgile, 2024-03-27:
### PyBaMM Version

24.1

### Python Version

3.10.14

### Describe the bug

When I use the official example code will generate errors. the casadi is 3.6.5. And I used to use it well before, but it hasn't worked out recently.
```
# load model and parameters
model = pybamm.lithium_ion.DFN()
param = model.default_parameter_values
param["Lower voltage cut-off [V]"] = 3.6

# load solvers
safe_solver = pybamm.CasadiSolver(atol=1e-3, rtol=1e-3, mode="safe")
fast_solver = pybamm.CasadiSolver(atol=1e-3, rtol=1e-3, mode="fast")

# create simulations
safe_sim = pybamm.Simulation(model, parameter_values=param, solver=safe_solver)
fast_sim = pybamm.Simulation(model, parameter_values=param, solver=fast_solver)

# solve
safe_sim.solve([0, 3600])
print(f"Safe mode solve time: {safe_sim.solution.solve_time}")
fast_sim.solve([0, 3600])
print(f"Fast mode solve time: {fast_sim.solution.solve_time}")

# plot solutions
pybamm.dynamic_plot([safe_sim, fast_sim])
```
[image]
[image]

But my casadi is not a problem.
[image]

### Steps to Reproduce

```
import pybamm
# load model and parameters
model = pybamm.lithium_ion.DFN()
param = model.default_parameter_values
param["Lower voltage cut-off [V]"] = 3.6

# load solvers
safe_solver = pybamm.CasadiSolver(atol=1e-3, rtol=1e-3, mode="safe")
fast_solver = pybamm.CasadiSolver(atol=1e-3, rtol=1e-3, mode="fast")

# create simulations
safe_sim = pybamm.Simulation(model, parameter_values=param, solver=safe_solver)
fast_sim = pybamm.Simulation(model, parameter_values=param, solver=fast_solver)

# solve
safe_sim.solve([0, 3600])
print(f"Safe mode solve time: {safe_sim.solution.solve_time}")
fast_sim.solve([0, 3600])
print(f"Fast mode solve time: {fast_sim.solution.solve_time}")

# plot solutions
pybamm.dynamic_plot([safe_sim, fast_sim])
```

tinosulzer, 2024-03-27:
Can you try one of casadi's ODE solve examples?

agriyakhetarpal, 2024-03-27:
This looks to be a similar bug as #3783

YoungAgile, 2024-03-27:
> Can you try one of casadi's ODE solve examples? Not using pybamm, using casadi directly to solve an ODE. And a casadi root-finding example since that seems to be the problem here

```
from casadi import *
x = MX.sym('x',2); # Two states

# Expression for ODE right-hand side
z = 1-x[1]**2
rhs = vertcat(z*x[0]-x[1],x[0])

ode = {}         # ODE declaration
ode['x']   = x   # states
ode['ode'] = rhs # right-hand side

# Construct a Function that integrates over 4s
F = integrator('F','cvodes',ode,{'tf':4})

# Start from x=[0;1]
res = F(x0=[0,1])

print(res["xf"])

# Sensitivity wrt initial state
res = F(x0=x)
S = Function('S',[x],[jacobian(res["xf"],x)])
print(S([0,1]))
```

[image]

agriyakhetarpal, 2024-03-27:
@YoungAgile, could you try re-installing CasADi with your preferred installation method? It looks like your CasADi installation is looking at the wrong path to find `casadi_integrator_cvodes.dll` (the `RuntimeError` is coming from `D:\bld\`, but your CasADi installation is present at `d:\anaconda3\envs\`?)

YoungAgile, 2024-03-28:
> @YoungAgile, could you try re-installing CasADi with your preferred installation method? It looks like your CasADi installation is looking at the wrong path to find `casadi_integrator_cvodes.dll` [...]

@agriyakhetarpal Thank you for your reply. I deleted the entire environment and reconfigured it. But I don't know why Casadi would search in the path `D: \ bld \` instead of the installed environment `d: \ anaconda3 \ envs\`

YoungAgile, 2024-03-28:
This issue seems to have occurred after I upgraded to anaconda.
Just now, I reset anaconda to its default settings, and this issue seems to have been resolved.
Thank you for your patient response. @agriyakhetarpal @tinosulzer 

[image]

agriyakhetarpal, 2024-03-28:
Thanks for the update, does this resolve the original error report with PyBaMM's `CasadiSolver` too?

agriyakhetarpal, 2024-03-28:
Thanks! I'm closing this issue – please feel free to reopen if you face a related issue or create another for an unrelated issue.

closed 2024-03-28 by agriyakhetarpal