pybamm-team/PyBaMM #3964: pybamm[bpx] migrate to pydantic v2

chuckliu1979, 2024-04-04:
### Description

the optional dependency 'bpx' 0.4.0 still use pydantic v1, it blocks others which depends on pydantic v2 to use pybamm.
I pulls a request for bpx project, and pybamm parameter_values create_from_bpx should be updated as well. 

### Motivation

i'm waiting https://github.com/FaradayInstitution/BPX/pull/51 review.  if approved and merged, create_from_bpx should be updated, accept both bpx <= 0.4.0 and new. 

### Possible Implementation

check bpx version, determine if need pydantic v2

kratman, 2024-04-04:
I will take a look at updating the version of BPX we use

kratman, 2024-05-03:
I am going to close this ticket for now. Since Pydantic v2 is not available in BPX yet and there is no new version of BPX, there is nothing that can be done. A new ticket can be opened after BPX gets updated

closed 2024-05-03 by kratman