pybamm-team/PyBaMM #3977: Reversible heat in what cases can be reflected in the results

Santiagopeacely, 2024-04-10:
I run this as follows, but the reversible heat of the result is 0. Why is that? If I wanted to embody reversible heat, how would I choose the additional model? Please correct me, thank you very much!
[image]
[image]

rtimms, 2024-04-10:
The "Chen2020" parameter set has 0 for the value of both entropic change coefficients. Try using another parameter set e.g. "ORegan2022". You can search for parameter values by doing `params.search("my search term")`.

Santiagopeacely, 2024-04-10:
Thank you very much for your answer. I wish you a happy life and smooth work

L Y?
***@***.***

抄送: "L ***@***.******@***.***&gt;;