pybamm-team/PyBaMM #3978: [Bug]: Can't use "CustomStepExplicit" and "CustomStepImplicit" in v24.1

WoohyunJeong95, 2024-04-10:
### PyBaMM Version

24.1

### Python Version

3.11.5

### Describe the bug

Hi, I'm trying the code in https://docs.pybamm.org/en/latest/source/examples/notebooks/simulations_and_experiments/custom-experiments.html

I'd like to define custom step with "CustomStepExplicit" or "CustomStepImplicit", but it gives me this attribute error. 
[image]

I also tried the code with colab (https://colab.research.google.com/github/pybamm-team/PyBaMM/blob/develop/docs/source/examples/notebooks/simulations_and_experiments/custom-experiments.ipynb), but it does not work as well.

I'd really appreciate it if you could help me with this.

Thank you,
Woohyun

### Steps to Reproduce

import pybamm

def custom_step_power(variables):
    target_power = 4
    voltage = variables["Voltage [V]"]
    return target_power / voltage

# Run for 10 minutes and plot
step = pybamm.step.CustomStepExplicit(custom_step_power, duration=600)
sol = pybamm.Simulation(
    model, experiment=step, parameter_values=parameter_values
).solve()
pybamm.QuickPlot(sol, ["Current [A]", "Voltage [V]", "Power [W]"]).plot(0)

agriyakhetarpal, 2024-04-10:
Hi, @WoohyunJeong95 – thanks for opening this issue. This feature to enable custom experiment steps was added in #3835 and is not released yet as a part of a stable version of PyBaMM. It will be available with PyBaMM v24.5 and its release candidates. If you wish to use the feature right away, you may install PyBaMM from sources or from the git trunk. An installation guide is available here, for this purpose: https://docs.pybamm.org/en/latest/source/user_guide/installation/index.html

WoohyunJeong95, 2024-04-11:
Thank you! I appreciate it.

closed 2024-04-11 by WoohyunJeong95