pybamm-team/PyBaMM #4022: Add "Surface temperature" submodel

valentinsulzer, 2024-04-18:
This model accounts for the fact that the surface temperature of a cell in a thermal chamber is typically different from the "set"/"ambient" temperature of the chamber.

Suggested steps to do this below, but open to discussion:
- Add a submodel `SurfaceAmbientTemperature` which sets surface temperature equal to ambient temperature
- Add another submodel `SurfaceTemperatureODE` which implements the equation from [Mohan et al](https://d1wqtxts1xzle7.cloudfront.net/69868813/j.jpowsour.2014.01.09720210918-12240-190ns8h-libre.pdf?1632013230=&response-content-disposition=inline%3B+filename%3DA_lumped_parameter_electro_thermal_model.pdf&Expires=1713470219&Signature=VVdyggn0EMZaZUDAhblutOENZ6faE0Wua0uoq71ebx912paMfn2O2-lJq8SUK1nwkZf7v5jF65dIUF03R~IKZo5LXZdYEOb48oPshUycdG0h3K4EHPaGW07CNjd6e~UFP2JyNjUDMAnfP5oZHD~8cr5MRoTbNi8WTfcJ8SSdc1nyOr4uKy5xevjQKW3D0cvAbu9gSnbVZC52qdwM6jeobcJLAx7AhMetR3x1r2lHd6Jrrz-vznm1JeczXl6EzEzMgqr2tLOcKO7GLhlUVYIDhdGawR76FJHLBNyhAEPXtEMq7IUibswa~~2fdgZNTemj9lCjsExrmrri-xZhHFL5vQ__&Key-Pair-Id=APKAJLOHF5GGSLRBV4ZA):
[image]
- In all other thermal submodels, replace "ambient temperature" with "surface temperature"
- In `BaseBatteryModel`, add an option for "surface temperature" which can be "ambient" (default) or "ODE", then add an if statement for selecting the appropriate model
- Add a test for selecting the "ODE" option
- Add docs

brosaplanella, 2024-04-20:
Would that be part of the thermal models or a new type of submodel?

valentinsulzer, 2024-04-21:
New type of submodel so that it is compatible with any thermal model.
One difficulty will be calculating the `T_s-T_c/R_c` term (surface cooling i.e. ambient heating) for models other than the lumped thermal model. See https://github.com/pybamm-team/PyBaMM/pull/4021#discussion_r1571171874

closed 2024-08-27 by kratman

wasitahir, 2024-09-16:
@valentinsulzer can the surface temperature model used for other chemistry parameters e.g., Prada2013?

PR #4203 (2024-08-27): Surface temperature model
# Description

From Lin et al 2014 (kudos @js1tr3 et al) 
[image]
[image]

Fixes #4022 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that speeds up the code)
- [ ] Bug fix (non-breaking change which fixes an issue)

# Key checklist:

- [ ] No styl [... cut]