pybamm-team/PyBaMM #4047: [Bug]: No module named 'ipywidgets'

swsyoon, 2024-04-26:
### PyBaMM Version

24.1

### Python Version

3.11.9

### Describe the bug

I am running tutorial 1 in jupyter lab (Version 4.1.6) at Miniforge (  [https://github.com/conda-forge/miniforge/releases/latest/download/Miniforge3-Windows-x86_64.exe](url) ) and error occurs when I execute `sim.plot()` . 
[image]

### Steps to Reproduce

Simply run the tutorial

### Relevant log output

```shell
---------------------------------------------------------------------------
ModuleNotFoundError                       Traceback (most recent call last)
Cell In[5], line 1
----> 1 sim.plot()

File ~\AppData\Local\miniforge-pypy3\envs\pybamm\Lib\site-packages\pybamm\simulation.py:1054, in Simulation.plot(self, output_variables, **kwargs)
   1051 if output_variables is None:
   1052     output_variables = self._output_variables
-> 1054 self.quick_plot = pybamm.dynamic_plot(
   1055     self._solution, output_variables=output_variables, **kwargs
   1056 )
   1058 return self.quick_plot

File ~\AppData\Local\miniforge-pypy3\envs\pybamm\Lib\site-packages\pybamm\plotting\dynamic_plot.py:21, in dynamic_plot(*args, **kwargs)
     19 kwargs_for_class = {k: v for k, v in kwargs.items() if k != "testing"}
     20 plot = pybamm.QuickPlot(*args, **kwargs_for_class)
---> 21 plot.dynamic_plot(kwargs.get("testing", False))
     22 return plot

File ~\AppData\Local\miniforge-pypy3\envs\pybamm\Lib\site-packages\pybamm\plotting\quick_plot.py:666, in QuickPlot.dynamic_plot(self, testing, step)
    653 """
    654 Generate a dynamic plot with a slider to control the time.
    655 
   (...)
    663 
    664 """
    665 if pybamm.is_notebook():  # pragma: no cover
--> 666     import ipywidgets as widgets
    668     step = step or self.max_t / 100
    669     widgets.interact(
    670         lambda t: self.plot(t, dynamic=False),
    671         t=widgets.FloatSlider(
   (...)
    674         continuous_update=False,
    675     )

ModuleNotFoundError: No module named 'ipywidgets'
```

kratman, 2024-04-26:
So in the first cell the install of pybamm with the `plot` dependencies is commented out. Probably just need to make sure those are installed. If that does not work then I will look into this further

swsyoon, 2024-04-26:
I had to comment out due to another error which does not seem to change the result. Here is a screenshot after removing the comment. The same error appears when execute `sim.plot()`

[image]

kratman, 2024-04-26:
It looks like the install failed with an error due to the comment after the -q flag, can you remove the comment in the first line and see if it installs correctly?

swsyoon, 2024-04-26:
Removing the comment after the -q flag solves the error at the line but still `sim.plot()` generates the same error
[image]

kratman, 2024-04-26:
Ok I will take a look then. Thanks for checking

agriyakhetarpal, 2024-04-26:
We can consider declaring `ipywidgets` as an optional dependency under `[plot]`.

agriyakhetarpal, 2024-04-26:
> Yeah still good to know why there is a difference with this setup

I found the problem – `pybamm.QuickPlot.dynamic_plot` requires it, but the import is hidden inside the `is_notebook()` function, and therefore PyBaMM does not fail. I think adding it to the `[plot]` dependencies and replacing the import with `import_optional_dependency()` should be alright (along with corresponding changes in the installation guide).

kratman, 2024-05-03:
So I checked on both pip and anaconda packages, ipywidgets is a dependency of Jupyter. This sounds like it might be an issue with the miniforge version of Jupyter lab rather than an issue with PyBaMM itself

agriyakhetarpal, 2024-05-03:
Thanks for checking – Miniforge is just the same as `conda`/`mamba` but tailored for the `conda-forge` channel rather than `defaults`. So, yes, the notebook should work without changes from our end.

closed 2024-05-04 by kratman