pybamm-team/PyBaMM #4048: [Bug]: Tutorial 8 - Solver options | Not showing what it intends to

ottarg, 2024-04-27:
### PyBaMM Version

24.1

### Python Version

3.12.2

### Describe the bug

When I run the tutorial 8 notebook, it runs one sim in CasADi safe mode, one in CasADi fast mode and then plots the two for comparison.

The text in the notebook indicates that the difference should be that the safe mode correctly stops at the given voltage cut-off and that fast mode continues until the end of the requested solution time. However, both simulations appear to stop at the cut-off voltage. 
[image]

I get the same behavior both in the google colab notebook as I get locally: 
[link to google colab notebook](https://colab.research.google.com/github/pybamm-team/PyBaMM/blob/main/docs/source/examples/notebooks/getting_started/tutorial-8-solver-options.ipynb)

When I change the script such that the fast solver does not get an altered parameter for cut-off voltage, it does complete the full hour simulation -- so it appears that it is indeed respecting the stop condition (which is counter to the description of what this mode does)
`# create simulations`
`safe_sim = pybamm.Simulation(model, parameter_values=param, solver=safe_solver)`
`fast_sim = pybamm.Simulation(model, solver=fast_solver)`

[image]

### Steps to Reproduce

Run the tutorial 8 notebook with no changes

kratman, 2024-05-03:
Hi @ottarg, the notebook does not force a crash for the fast solver to make sure everything runs.

The description actually focuses on them giving the same exact result rather than showing that it stops at the cutoff:
```
We see that both solvers give the same solution and that the "fast" solver, as the name suggests,
runs faster. However, if the simulation time was longer the "fast" solver would not notice that the
battery is discharging beyond its cut-off voltage and the solver would crash.
```

I think that there could be another plot to show that the safe solver stops, but as it is setup now, it is not a bug

kratman, 2024-05-03:
I am going to close this for now since it is behaving as intended.

closed 2024-05-03 by kratman