pybamm-team/PyBaMM #4058: Calculate summary variables only when they are called

rtimms, 2024-05-01:
As discussed in #3995 

- create a `SummaryVariable` object that saves `solution.last_state` and uses it to calculate the esoh variables only when they are called (the same way as `Solution` calculates the variables only when they are called)
- allow users to pass in functions that calculate summary variables based on other variables

pipliggins, 2024-11-27:
@rtimms reading through #3995 it looks like the decision was to not allow users to define their own summary variables, have I got that right?

rtimms, 2024-11-28:
If we just keep summary variables as a list then I think defining your own summary variables is hard. Right now the summary variables just calculate the value of the variable at the last time point, and then change in that variable. 

We could do something more complicated like pass in functions that take a solution and perform some operation to get a summary variable, but I'm not sure there is a strong use case for this. It can always be done as a post-processing step. If you only want the last value of some quantity then users can add custom variables to the model and then add that to the list of summary variables.

PR #4621 (2024-12-19): Summary variables calculated only when called
# Description

Rather than calculating summary variables and storing them in every Solution class, a 'SummaryVariables' class is created which stores the first + last state of the solution, plus any relevant esoh parameters, so that the summary variables are only calculated and stored when they are called (similar to accessing standard variables). If an esoh variable is requested, they are all calculated and stored at once to reduce the number of calls to the esoh solver.

rather than viewing the list of summary variable names as
```
sol.summary_va [... cut]
closed 2024-12-19 by martinjrobins