pybamm-team/PyBaMM #4060: [Bug]: Nominal capacity : Parameter Change

Not-slytherin, 2024-05-03:
### PyBaMM Version

23.4.1

### Python Version

3.11.5

### Describe the bug

I'm currently simulating a Pouch-Cell with a nominal capacity of 65Ah. When I run the experiment using the predefined nominal cell capacity with OKane2022, everything works fine. However, when I adjust it to match the specific cell I'm working with, I encounter a debugging message.

Can you please indicate to me what may be the problem, and thank you for prividing the community with this very helpfull tool.

### Steps to Reproduce

Action that created the error:
# Parameters Change
parameter_values["Nominal cell capacity [A.h]"] = 65

Error:
The linesearch algorithm failed with too small a step.
At t = 50.7015 and h = 1.02133e-09, the corrector convergence failed repeatedly or with |h| = hmin.
The linesearch algorithm failed with too small a step.
The linesearch algorithm failed with too small a step.
The linesearch algorithm failed with too small a step.
The linesearch algorithm failed with too small a step.
2024-05-03 17:10:16.280 - [ERROR] callbacks.on_experiment_error(224): Simulation error: Maximum number of decreased steps occurred at t=0.0 (final SolverError: 'Error in Function::call for 'F' [IdasInterface] at .../casadi/core/function.cpp:1401:
Error in Function::call for 'F' [IdasInterface] at .../casadi/core/function.cpp:330:
.../casadi/interfaces/sundials/idas_interface.cpp:596: IDACalcIC returned "IDA_LINESEARCH_FAIL". Consult IDAS documentation.'). For a full solution try reducing dt_max (currently, dt_max=600) and/or reducing the size of the time steps or period of the experiment. Set `return_solution_if_failed_early=True` to return the solution object up to the point where failure occured.

Unexpected exception formatting exception. Falling back to standard exception

Traceback (most recent call last):
  File "C:\Users\OO000009\AppData\Local\anaconda3\Lib\site-packages\pybamm\solvers\casadi_solver.py", line 679, in _run_integrator
    pybamm.logger.debug("Calling casadi integrator")
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\OO000009\AppData\Local\anaconda3\Lib\site-packages\casadi\casadi.py", line 23390, in __call__
    return self.call(kwargs)
           ^^^^^^^^^^^^^^^^^
  File "C:\Users\OO000009\AppData\Local\anaconda3\Lib\site-packages\casadi\casadi.py", line 20039, in call
    return _casadi.Function_call(self, *args)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
RuntimeError: Error in Function::call for 'F' [IdasInterface] at .../casadi/core/function.cpp:1401:
Error in Function::call for 'F' [IdasInterface] at .../casadi/core/function.cpp:330:
.../casadi/interfaces/sundials/idas_interface.cpp:596: IDACalcIC returned "IDA_LINESEARCH_FAIL". Consult IDAS documentation.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\OO000009\AppData\Local\anaconda3\Lib\site-packages\pybamm\solvers\casadi_solver.py", line 248, in _integrate
    f"{t_window[0]:.2f} < t < {t_window[-1]:.2f}"
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\OO000009\AppData\Local\anaconda3\Lib\site-packages\pybamm\solvers\casadi_solver.py", line 685, in _run_integrator
    # If it doesn't work raise error
pybamm.expression_tree.exceptions.SolverError: Error in Function::call for 'F' [IdasInterface] at .../casadi/core/function.cpp:1401:
Error in Function::call for 'F' [IdasInterface] at .../casadi/core/function.cpp:330:
.../casadi/interfaces/sundials/idas_interface.cpp:596: IDACalcIC returned "IDA_LINESEARCH_FAIL". Consult IDAS documentation.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\OO000009\AppData\Local\anaconda3\Lib\site-packages\IPython\core\interactiveshell.py", line 3526, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "C:\Users\OO000009\AppData\Local\Temp\ipykernel_11516\1774507600.py", line 51, in <module>
    solution = sim.solve(initial_soc=0.02)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\OO000009\AppData\Local\anaconda3\Lib\site-packages\pybamm\simulation.py", line 807, in solve
  File "C:\Users\OO000009\AppData\Local\anaconda3\Lib\site-packages\pybamm\simulation.py", line 785, in solve
    if cycle_num == 1 and step_num == 1:
                ^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\OO000009\AppData\Local\anaconda3\Lib\site-packages\pybamm\solvers\base_solver.py", line 1235, in step
    final_event_values = {}
       ^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\OO000009\AppData\Local\anaconda3\Lib\site-packages\pybamm\solvers\casadi_solver.py", line 283, in _integrate
    else:
        ^
pybamm.expression_tree.exceptions.SolverError: Maximum number of decreased steps occurred at t=0.0 (final SolverError: 'Error in Function::call for 'F' [IdasInterface] at .../casadi/core/function.cpp:1401:
Error in Function::call for 'F' [IdasInterface] at .../casadi/core/function.cpp:330:
.../casadi/interfaces/sundials/idas_interface.cpp:596: IDACalcIC returned "IDA_LINESEARCH_FAIL". Consult IDAS documentation.'). For a full solution try reducing dt_max (currently, dt_max=600) and/or reducing the size of the time steps or period of the experiment. Set `return_solution_if_failed_early=True` to return the solution object up to the point where failure occured.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\OO000009\AppData\Local\anaconda3\Lib\site-packages\IPython\core\interactiveshell.py", line 2120, in showtraceback
    stb = self.InteractiveTB.structured_traceback(
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\OO000009\AppData\Local\anaconda3\Lib\site-packages\IPython\core\ultratb.py", line 1435, in structured_traceback
    return FormattedTB.structured_traceback(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\OO000009\AppData\Local\anaconda3\Lib\site-packages\IPython\core\ultratb.py", line 1326, in structured_traceback
    return VerboseTB.structured_traceback(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\OO000009\AppData\Local\anaconda3\Lib\site-packages\IPython\core\ultratb.py", line 1173, in structured_traceback
    formatted_exception = self.format_exception_as_a_whole(etype, evalue, etb, number_of_lines_of_context,
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\OO000009\AppData\Local\anaconda3\Lib\site-packages\IPython\core\ultratb.py", line 1088, in format_exception_as_a_whole
    frames.append(self.format_record(record))
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\OO000009\AppData\Local\anaconda3\Lib\site-packages\IPython\core\ultratb.py", line 970, in format_record
    frame_info.lines, Colors, self.has_colors, lvals
    ^^^^^^^^^^^^^^^^
  File "C:\Users\OO000009\AppData\Local\anaconda3\Lib\site-packages\IPython\core\ultratb.py", line 792, in lines
    return self._sd.lines
           ^^^^^^^^^^^^^^
  File "C:\Users\OO000009\AppData\Local\anaconda3\Lib\site-packages\stack_data\utils.py", line 145, in cached_property_wrapper
    value = obj.__dict__[self.func.__name__] = self.func(obj)
                                               ^^^^^^^^^^^^^^
  File "C:\Users\OO000009\AppData\Local\anaconda3\Lib\site-packages\stack_data\core.py", line 698, in lines
    pieces = self.included_pieces
             ^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\OO000009\AppData\Local\anaconda3\Lib\site-packages\stack_data\utils.py", line 145, in cached_property_wrapper
    value = obj.__dict__[self.func.__name__] = self.func(obj)
                                               ^^^^^^^^^^^^^^
  File "C:\Users\OO000009\AppData\Local\anaconda3\Lib\site-packages\stack_data\core.py", line 649, in included_pieces
    pos = scope_pieces.index(self.executing_piece)
                             ^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\OO000009\AppData\Local\anaconda3\Lib\site-packages\stack_data\utils.py", line 145, in cached_property_wrapper
    value = obj.__dict__[self.func.__name__] = self.func(obj)
                                               ^^^^^^^^^^^^^^
  File "C:\Users\OO000009\AppData\Local\anaconda3\Lib\site-packages\stack_data\core.py", line 628, in executing_piece
    return only(
           ^^^^^
  File "C:\Users\OO000009\AppData\Local\anaconda3\Lib\site-packages\executing\executing.py", line 164, in only
    except (SyntaxError, ValueError):
executing.executing.NotOneValueFound: Expected one value, found 0

kratman, 2024-05-03:
Hi @Not-slytherin, The nominal capacity is part of the parameterization of the cell model itself. Changing from the capacity of 5 Ah in the model to 65 Ah in your desired simulation would require changing more than just the capacity value. Things like the cell/electrode dimensions and lithium concentration would need to be changed as well.

PyBaMM does not go through and adjust all the other parameters to match the nominal capacity. Unfortunately this is not trivial to accomplish

Not-slytherin, 2024-05-06:
Thank you so much for the clarification.

Best regards,

kratman, 2024-05-06:
If you are interested in modeling larger batteries, you can look into liionpack which lets you turn a bunch of cells in PyBaMM into a single battery

closed 2024-05-06 by kratman