pybamm-team/PyBaMM #4075: [Bug]:

Not-slytherin, 2024-05-08:
### PyBaMM Version

23.4.1

### Python Version

3.11.5

### Describe the bug

I wanted to perform a simulation for a pouch cell and based on the provided example of the model i just simply run the code but it keeps on giving the following error.
I also have a problem on the other hand when i change the capacity of the cell from 5 to 65 Ah. it gives a debug and i know it's because i should also change gemetrical parameters besides concentrations and so on. So,  knowing that all the parameters set are for cylindrical cells, can you please indicate to me where i can find Electrochemical model parameters of a pouch cell so i can get close and help the model perform the simulation with less excpected errors.

Thank you for your understanding,

### Steps to Reproduce

import pybamm
import matplotlib.pyplot as plt
import numpy as np

param= pybamm.ParameterValues("OKane2022")
dfn = pybamm.lithium_ion.DFN(    {"current collector": "potential pair", "dimensionality": 2, "thermal": "x-lumped"},

)
npts = 16
var_pts = {
    "x_n": npts,
    "x_s": npts,
    "x_p": npts,
    "r_n": npts,
    "r_p": npts,
    "z": npts,
}
experiment = pybamm.Experiment(
    [
        (
            "Charge at 0.5 C for 4 hours or until 4.2 V",
            "Rest for 30 minutes",
            "Discharge at 0.5 C for 4 hours or until 2.5 V",
            "Rest for 7 hours",
        ),
    ]
    * 3
    + [
        (
            "Charge at 26 A for 2 hours or until 4.2 V",
        )
    ]
)
# Create a simulation with the model and experiment
sim = pybamm.Simulation(model=dfn, experiment=experiment, parameter_values=param, var_pts=var_pts)

# Solve the simulation
sim.solve()

### Relevant log output

```shell
The followin erros is the response of the simulation code provided:

---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
File ~\AppData\Local\anaconda3\Lib\site-packages\pybamm\parameters\parameter_values.py:603, in ParameterValues.process_symbol(self, symbol)
    602 try:
--> 603     return self._processed_symbols[symbol]
    604 except KeyError:

KeyError: Division(0x82d8dea9f08b64c, /, children=['laplacian(X-averaged cell temperature [K]) + mass(X-averaged cell temperature [K]) @ ((Negative current collector thickness [m] * Negative current collector conductivity [S.m-1] * grad squared(Negative current collector potential [V]) + (Negative electrode thickness [m] + Separator thickness [m] + Positive electrode thickness [m]) * (((Negative electrode thickness [m] * x-average(-(-Negative electrode conductivity [S.m-1] * (Negative electrode active material volume fraction ** Negative electrode Bruggeman coefficient (electrode)) * grad(Negative electrode potential [V])) inner product grad(Negative electrode potential [V])) + Positive electrode thickness [m] * x-average(-(-Positive electrode conductivity [S.m-1] * (Positive electrode active material volume fraction ** Positive electrode Bruggeman coefficient (electrode)) * grad(Positive electrode potential [V])) inner product grad(Positive electrode potential [V]))) / (Negative electrode thickness [m] + Separator thickness [m] + Positive electrode thickness [m])) - x-average((Electrolyte conductivity [S.m-1] * concatenation(Negative electrode porosity ** Negative electrode Bruggeman coefficient (electrolyte), Separator porosity ** Separator Bruggeman coefficient (electrolyte), Positive electrode porosity ** Positive electrode Bruggeman coefficient (electrolyte)) * (((broadcast(Ideal gas constant [J.K-1.mol-1] * X-averaged cell temperature [K] / Faraday constant [C.mol-1]) * (2.0 - (2.0 * Cation transference number)) * Thermodynamic factor / (maximum(Porosity times concentration [mol.m-3](Negative electrode porosity times concentration [mol.m-3], Separator porosity times concentration [mol.m-3], Positive electrode porosity times concentration [mol.m-3]) / concatenation(Negative electrode porosity, Separator porosity, Positive electrode porosity), 0.01))) * grad(Porosity times concentration [mol.m-3](Negative electrode porosity times concentration [mol.m-3], Separator porosity times concentration [mol.m-3], Positive electrode porosity times concentration [mol.m-3]) / concatenation(Negative electrode porosity, Separator porosity, Positive electrode porosity))) - grad(Electrolyte potential [V](Negative electrolyte potential [V], Separator electrolyte potential [V], Positive electrolyte potential [V])))) inner product grad(Electrolyte potential [V](Negative electrolyte potential [V], Separator electrolyte potential [V], Positive electrolyte potential [V]))) + (Negative electrode thickness [m] * x-average((3.0 * Negative electrode active material volume fraction / Negative particle radius [m]) * 2.0 * Negative electrode exchange-current density [A.m-2] * sinh(0.5 * Faraday constant [C.mol-1] * (Negative electrode potential [V] - Negative electrolyte potential [V] - (Negative electrode OCP [V] + 1e-06 * (1.0 / (maximum(minimum(boundary value(Negative particle concentration [mol.m-3]) / Maximum concentration in negative electrode [mol.m-3], 0.9999999999), 1e-10)) + 1.0 / (-1.0 + maximum(minimum(boundary value(Negative particle concentration [mol.m-3]) / Maximum concentration in negative electrode [mol.m-3], 0.9999999999), 1e-10))) + broadcast(X-averaged cell temperature [K] - Reference temperature [K]) * Negative electrode OCP entropic change [V.K-1])) / broadcast(Ideal gas constant [J.K-1.mol-1] * X-averaged cell temperature [K])) * (Negative electrode potential [V] - Negative electrolyte potential [V] - (Negative electrode OCP [V] + 1e-06 * (1.0 / (maximum(minimum(boundary value(Negative particle concentration [mol.m-3]) / Maximum concentration in negative electrode [mol.m-3], 0.9999999999), 1e-10)) + 1.0 / (-1.0 + maximum(minimum(boundary value(Negative particle concentration [mol.m-3]) / Maximum concentration in negative electrode [mol.m-3], 0.9999999999), 1e-10))) + broadcast(X-averaged cell temperature [K] - Reference temperature [K]) * Negative electrode OCP entropic change [V.K-1]))) + Positive electrode thickness [m] * x-average((3.0 * Positive electrode active material volume fraction / Positive particle radius [m]) * 2.0 * Positive electrode exchange-current density [A.m-2] * sinh(0.5 * Faraday constant [C.mol-1] * (Positive electrode potential [V] - Positive electrolyte potential [V] - (Positive electrode OCP [V] + 1e-06 * (1.0 / (maximum(minimum(boundary value(Positive particle concentration [mol.m-3]) / Maximum concentration in positive electrode [mol.m-3], 0.9999999999), 1e-10)) + 1.0 / (-1.0 + maximum(minimum(boundary value(Positive particle concentration [mol.m-3]) / Maximum concentration in positive electrode [mol.m-3], 0.9999999999), 1e-10))) + broadcast(X-averaged cell temperature [K] - Reference temperature [K]) * Positive electrode OCP entropic change [V.K-1])) / broadcast(Ideal gas constant [J.K-1.mol-1] * X-averaged cell temperature [K])) * (Positive electrode potential [V] - Positive electrolyte potential [V] - (Positive electrode OCP [V] + 1e-06 * (1.0 / (maximum(minimum(boundary value(Positive particle concentration [mol.m-3]) / Maximum concentration in positive electrode [mol.m-3], 0.9999999999), 1e-10)) + 1.0 / (-1.0 + maximum(minimum(boundary value(Positive particle concentration [mol.m-3]) / Maximum concentration in positive electrode [mol.m-3], 0.9999999999), 1e-10))) + broadcast(X-averaged cell temperature [K] - Reference temperature [K]) * Positive electrode OCP entropic change [V.K-1])))) / (Negative electrode thickness [m] + Separator thickness [m] + Positive electrode thickness [m]) + (Negative electrode thickness [m] * x-average((3.0 * Negative electrode active material volume fraction / Negative particle radius [m]) * 2.0 * Negative electrode exchange-current density [A.m-2] * sinh(0.5 * Faraday constant [C.mol-1] * (Negative electrode potential [V] - Negative electrolyte potential [V] - (Negative electrode OCP [V] + 1e-06 * (1.0 / (maximum(minimum(boundary value(Negative particle concentration [mol.m-3]) / Maximum concentration in negative electrode [mol.m-3], 0.9999999999), 1e-10)) + 1.0 / (-1.0 + maximum(minimum(boundary value(Negative particle concentration [mol.m-3]) / Maximum concentration in negative electrode [mol.m-3], 0.9999999999), 1e-10))) + broadcast(X-averaged cell temperature [K] - Reference temperature [K]) * Negative electrode OCP entropic change [V.K-1])) / broadcast(Ideal gas c [... cut ...]
     79         raise KeyError(
     80             f"'{key}' not found. Use the dimensional version '{k}' instead."
     81         )
---> 82 raise KeyError(f"'{key}' not found. Best matches are {best_matches}")

KeyError: "'Negative current collector surface heat transfer coefficient [W.m-2.K-1]' not found. Best matches are ['Negative current collector thermal conductivity [W.m-1.K-1]', 'Negative current collector specific heat capacity [J.kg-1.K-1]', 'Total heat transfer coefficient [W.m-2.K-1]']"
```

kratman, 2024-05-15:
@Not-slytherin Sorry for the slow response.

Unfortunately not all of the models are compatible with every option in PyBaMM. In this case your 2D simulation with x-lumped thermal requires the parameter: `"Negative current collector surface heat transfer coefficient [W.m-2.K-1]"`, which is only properly defined in `Ecker2015`. A bunch of the other parameter sets define that variable to be zero, which effectively removes that aspect of the thermal model.

You could set those parameters to be zero with the understanding that the thermal analysis may be incomplete, or you can change the setup.

For example the following should work for a pouch cell with OKane2022:
```python
import pybamm

param = pybamm.ParameterValues("OKane2022")
dfn = pybamm.lithium_ion.DFN({"cell geometry": "pouch", "thermal": "lumped"})
experiment = pybamm.Experiment(
    [
        (
            "Charge at 0.5 C for 4 hours or until 4.2 V",
            "Rest for 30 minutes",
            "Discharge at 0.5 C for 4 hours or until 2.5 V",
            "Rest for 7 hours",
        ),
    ] * 3
        +
    [
        (
            "Charge at 26 A for 2 hours or until 4.2 V",
        )
    ]
)
sim = pybamm.Simulation(model=dfn, experiment=experiment, parameter_values=param)
sim.solve()
```

This might not be the exact geometry/configuration you want, but gets you started.

kratman, 2024-05-15:
Let me know if that helps