pybamm-team/PyBaMM #4079: Library is useless and needs to be a simulation entity (think Simulink)

igbokwedaniel, 2024-05-12:
### Description

I have spent the better part of the last month unsuccessfully trying to integrate PyBAMM in an actual use case scenario of a off grid electrical system. In "REAL LIFE" no one jut cycles batteries or shoves in static arbitrary current fuction. real systems must constantly evaluate SOC, voltage level and determin wether to draw power or charge or do anything. PyBAMM at this point seems like something that was done to validate theoretical work as it cannnot be coupled sensibly to battery powered/aided systems. 

### Motivation

A battery model is exceedingly useless if it cannot be coupled to other simulation models of larger systems. most battery use cases especially for degradation is not based off wanton laboratory cycling as is the vision  of the engineers working on PyBAMM

### Possible Implementation

the current  CustomStepExplicit is incredibly useless when considering that any real life external circuit must constantly monitor OCV/SOC to determine I_out after extensive circuit calculations. a true arbitrary current function must be established for real life use cases to be simulated with PyBAMM else we might as well all crawl back to simulink/COMSOL and open source be damned

brosaplanella, 2024-05-13:
Hi, thanks for your feedback. The comments, however are quite broad and not actionable, so closing this issue. If you could narrow down your comments into some specific and actionable suggestions and open a separate issue for them that would be helpful.

closed 2024-05-13 by brosaplanella

igbokwedaniel, 2024-05-13:
> Thanks for your feedback. I'm sorry it took you a month to realize pybamm was not built for your use case. Would you like a refund?

Lol. typical academic, perfectly content with his intellectually superior yet unusuable "invention"... why dont we just cycle a battery with 5C 100 time, because that's what real life battery usage looks like... do you even hear yourself?..... the Battery revolution continues!!!! idiot

igbokwedaniel, 2024-05-13:
> Thanks for your feedback. I'm sorry it took you a month to realize pybamm was not built for your use case. Would you like a refund?

Oh  my God, you're a genius my very good sir. I worship your intellectual eminence