pybamm-team/PyBaMM #4080: [Bug]: thermal option does not work in 2 phase dfn model

llh929423719, 2024-05-13:
### PyBaMM Version

24.1

### Python Version

3.9

### Describe the bug

thermal option does not work in 2 phase dfn model

### Steps to Reproduce

model = pybamm.lithium_ion.DFN(
    {
        "particle phases": ("2", "1"),
        "open-circuit potential": (("single", "current sigmoid"), "single"),
        "thermal": "lumped",
        "cell geometry": "pouch",
    }
)

### Relevant log output

```shell
ModelError: Missing variable for submodel 'thermal': 'Negative electrode primaryvolumetric interfacial current density [A.m-3]'.
Check the selected submodels provide all of the required variables.
```

llh929423719, 2024-05-13:
Adding a space after {phase} in base_thermal.py  _get_standard_coupled_variables() seems to solve the problem

rtimms, 2024-05-13:
Thanks, the way to fix this is to replace [this line](https://github.com/pybamm-team/PyBaMM/blob/1dae0bddd25c2b998ef6b635313c7bd0e6d8a33c/pybamm/models/submodels/thermal/base_thermal.py#L141) with 

`            phase_names = ["primary ", "secondary "]`

so that you don't end up with an extra space in the single phase model. Could you open a PR with a fix?

kratman, 2024-05-13:
I will do a quick patch for this so we can get this into the release

llh929423719, 2024-05-18:
> Thanks, the way to fix this is to replace [this line](https://github.com/pybamm-team/PyBaMM/blob/1dae0bddd25c2b998ef6b635313c7bd0e6d8a33c/pybamm/models/submodels/thermal/base_thermal.py#L141) with [...]

Thanks, it does work

closed 2024-05-18 by llh929423719

kratman, 2024-05-18:
@llh929423719 This should be in the next release of pybamm, I just need to write a test then we will get it merged