pybamm-team/PyBaMM #4082: Documentation fails to build in CI

brosaplanella, 2024-05-13:
The documentation fails to build in the CI for Ubuntu and Python 3.12: https://github.com/pybamm-team/PyBaMM/actions/runs/9056632702/job/24879466141

Not sure what the issue is, but seems to be failing repeatedly, while working for other Python versions.

agriyakhetarpal, 2024-05-13:
Ah, looks like we will still require the `session.install("setuptools")` hack for `pybtex` to work on Python 3.12. We have not been testing the documentation builds with 3.12, but with 3.11, rather, in the regular PR tests. As an aside, we should look to revise the workflows so that the nightly tests also run in parallel rather than running the different sessions sequentially.

cringeyburger, 2024-05-16:
Hi! I would like to work on this issue

arjxn-py, 2024-05-16:
Assigned this to you @cringeyburger, feel free to let know in case you'd want to discuss something related.

cringeyburger, 2024-05-18:
From what I understand, I need to add a temporary fix to `noxfile` by adding `session.install("setuptools", silent=False)`?

agriyakhetarpal, 2024-05-18:
> From what I understand, I need to add a temporary fix to `noxfile` by adding `session.install("setuptools", silent=False)`?

In the docs session, yes, that – and then parallelise the scheduled tests to resemble the `test_on_push.yml` job minus all of the caching stuff.

PR #4102 (2024-05-22): Issue 4082 Documentation CI Fix
# Description

Please include a summary of the change and which issue is fixed. Please also include relevant motivation and context. List any dependencies that are required for this change.
The documentation built on CI fails with an error. It was found that we still need to install setuptools before docs/doctests sessions to avoid erroring out. 
Also, the scheduled jobs are parallelized for faster CI

Fixes #4082 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to d [... cut]
closed 2024-05-22 by agriyakhetarpal