pybamm-team/PyBaMM #4096: How to run a single charging simulation

Tangzl922, 2024-05-16:
Hello,

I want to simulate the charging and discharging process separately without using experiments. Is it possible to run a single charging simulation using only 'Simulation' class? I tried setting initial SOC but it kept discharging from that SOC.

Best regards

kratman, 2024-05-18:
It is possible to manipulate things like the current without using experiments directly as long as you are OK with more limited functionality.

A normal discharging simulation looks like this:
```
model = pybamm.lithium_ion.DFN()
parameter_values = pybamm.ParameterValues("Chen2020")
sim = pybamm.Simulation(model, parameter_values=parameter_values)
sim.solve([0, 3600], initial_soc=1.0)
sim.plot(["Current [A]", "Voltage [V]"])
```
[image]

You can change the state of charge and reverse the current to do charging instead:
```
model = pybamm.lithium_ion.DFN()
parameter_values = pybamm.ParameterValues("Chen2020")
parameter_values["Current function [A]"] *= -1
sim = pybamm.Simulation(model, parameter_values=parameter_values)
sim.solve([0, 3600], initial_soc=0.0)
sim.plot(["Current [A]", "Voltage [V]"])
```
[image]

Of course realistic charging and discharging will be slightly different, but this is a quick way it can be done.

kratman, 2024-05-18:
@Tangzl922 Let me know if that helps

Tangzl922, 2024-05-22:
Hi it works, thanks for your answer!

closed 2024-05-22 by kratman