pybamm-team/PyBaMM #4099: Run unit tests and integration tests together to speed up PR builds

agriyakhetarpal, 2024-05-17:
## Description

The PR builds run unit tests and integration tests separately right now, which leads to a lot of runner requests from GitHub Actions (more than what are available to us concurrently with a non-team or non-enterprise plan), and as a result they usually take up to an hour to complete. A way to improve the PR build times would be to run `nox -s unit` and `nox -s integration` together till the time #3617 is finished and the integration tests are migrated to support `pytest` invocations, because the `unit` tests after #3857 run in parallel, and usually pass within ~5-6 minutes across platforms (and are much faster within the range of <3 minutes on larger runners such as the `macos-14` ones).

## Proposed implementation

Modify `test_on_push.yml` to run `nox -s unit` and `nox -s integration` in the same step, remove one (either) of the redundant unit or the integration jobs, and adjust the job names as necessary. 

## Additional context

Discussed in the community bonding GSoC meeting on 17/05/2024 with @Saransh-cpp and @arjxn-py – cc: @prady0t (you'll have to comment on the issue in order for us to assign you to this)

prady0t, 2024-05-17:
I'll start working on this right away. You can assign this to me.

agriyakhetarpal, 2024-05-17:
Adding xrefs: #3662

PR #4108 (2024-05-27): Running tests together on github-actions.
# Description

Running unit and integration tests together for speedup.

Fixes #4099
closed 2024-05-27 by kratman