pybamm-team/PyBaMM #4113: Make function for handling time or reuse an existing one

kratman, 2024-05-28:
The following code was added in #4073:
```python
            elif term.endswith("s"):
                end_time_s = term.split("s")[0]
                termination_dict["time"] = (float(end_time_s), "s")
            elif term.endswith("min"):
                end_time_s = term.split("min")[0]
                termination_dict["time"] = (float(end_time_s) * 60, "s")
            elif term.endswith("h"):
                end_time_s = term.split("h")[0]
                termination_dict["time"] = (float(end_time_s) * 3600, "s")
```
This should probably be converted to a function. For example:

kratman, 2024-05-28:
Note: Part of this ticket is finding out if there are other instances where this change can be made. So whoever takes this up, please search for other time conversions

valentinsulzer, 2024-05-28:
There is a `_convert_time_to_seconds` function https://github.com/pybamm-team/PyBaMM/blob/develop/pybamm/experiment/step/base_step.py#L344

agriyakhetarpal, 2024-05-28:
From https://github.com/pybamm-team/PyBaMM/pull/4073#discussion_r1617775872 I think you'll need to comment here, @mleot, so that GitHub can add your name to the list of potential assignees

mleot, 2024-05-28:
Got it, commenting here.

PR #4209 (2024-07-01): Make function for handling time or reuse an existing one
Fixes #4113

# Description

Please include a summary of the change and which issue is fixed. Please also include relevant motivation and context. List any dependencies that are required for this change.

Fixes #4113

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functi [... cut]
commit deca7e2b (2024-07-01): Make function for handling time or reuse an existing one (#4209)

* Make function for handling time or reuse an existing one
Fixes #4113

* style: pre-commit fixes

* fixing raises RegEx Error Experiment Test

* fix pre-commit error

* Update pybamm/callbacks.py

---------
closed 2024-07-01 by kratman