pybamm-team/PyBaMM #4127: [Bug]: Minimum JAX version is 0.4.16

DavidMStraub, 2024-06-03:
### PyBaMM Version

24.1

### Python Version

3.10.12

### Describe the bug

When a JAX version between 0.4.0 and 0.4.15 is installed and PyBaMM is installed without the `[jax]` optional (which can easily happen e.g. in an environment where `pybamm[all]==24.1` is issued after `pybamm[jax]==23.9` has been issued before), PyBaMM cannot be imported because `pybamm.have_jax()` only checks for a version greater than `JAX_VERSION = "0.4"` (so, returns `True`), but the import `from jax.extend import linear_util as lu` only starts working with JAX >= 0.4.16. (See #3671 where this was changed to silence a deprecation warning.)

I suggest to add a minimum JAX version (currently 0.4.16) and compare to that in `have_jax`.

### Steps to Reproduce

```bash
python -m venv myenv
. myenv/bin/activate
pip install wheel
pip install jax==0.4.13 jaxlib==0.4.13 pybamm[all]==24.1
python -c 'import pybamm'
```

### Relevant log output

```shell
Traceback (most recent call last):                                                                                        File "<string>", line 1, in <module>                                                                                    File "/home/straub/myenv/lib/python3.10/site-packages/pybamm/__init__.py", line 221, in <module>                          from .solvers.jax_bdf_solver import jax_bdf_integrate                                                                 File "/home/straub/myenv/lib/python3.10/site-packages/pybamm/solvers/jax_bdf_solver.py", line 13, in <module>             from jax.extend import linear_util as lu                                                                            ImportError: cannot import name 'linear_util' from 'jax.extend' (/home/straub/myenv/lib/python3.10/site-packages/jax/extend/__init__.py)
```

arjxn-py, 2024-06-03:
Thanks @DavidMStraub for opening this, will look into this

closed 2024-06-03 by agriyakhetarpal

DavidMStraub, 2024-06-03:
Thanks for the quick fix! Yes, I think this will work. However, I am wondering whether you should also have `==` instead of `>=` here in this case: https://github.com/pybamm-team/PyBaMM/blob/develop/pybamm/util.py#L357-L358

PR #4129 (2024-06-03): #4127 Set minimum `JAX==0.4.27`
# Description

Set minimum JAX version to `0.4.14` instead of `0.4`

Fixes #4127 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that speeds up the code)
- [x] Bug fix (non-breaking change which fixes an issue)

# Key checklist:

- [ ] No styl [... cut]