pybamm-team/PyBaMM #4135: Example " Stepping the solver"

Tangzl922, 2024-06-05:
Hello,

while running the example "stepping the solver" I got the different answer. The output variable would be repeated at each time step. I am using the newest version if you need the information. Do you have an idea why this happens?

```
dt = 360
time = 0
end_time = 3524

model = pybamm.lithium_ion.SPMe()
step_simulation = pybamm.Simulation(model)
while time < end_time:
    step_solution = step_simulation.step(dt)
    print("Time", time)
    print(step_solution["Voltage [V]"].entries)
    time += dt
```

And I got:

```
Time 0
[3.77048098 3.71252352]
Time 360
[3.77048098 3.71252352 3.71252352 3.68216967]
Time 720
[3.77048098 3.71252352 3.71252352 3.68216967 3.68216967 3.66127177]
Time 1080
[3.77048098 3.71252352 3.71252352 3.68216967 3.68216967 3.66127177
 3.66127177 3.64333974]
Time 1440
[3.77048098 3.71252352 3.71252352 3.68216967 3.68216967 3.66127177
 3.66127177 3.64333974 3.64333974 3.61170673]
Time 1800
[3.77048098 3.71252352 3.71252352 3.68216967 3.68216967 3.66127177
 3.66127177 3.64333974 3.64333974 3.61170673 3.61170673 3.59710873]
Time 2160
[3.77048098 3.71252352 3.71252352 3.68216967 3.68216967 3.66127177
 3.66127177 3.64333974 3.64333974 3.61170673 3.61170673 3.59710873
 3.59710873 3.58822263]
Time 2520
[3.77048098 3.71252352 3.71252352 3.68216967 3.68216967 3.66127177
 3.66127177 3.64333974 3.64333974 3.61170673 3.61170673 3.59710873
 3.59710873 3.58822263 3.58822263 3.58057678]
Time 2880
[3.77048098 3.71252352 3.71252352 3.68216967 3.68216967 3.66127177
 3.66127177 3.64333974 3.64333974 3.61170673 3.61170673 3.59710873
 3.59710873 3.58822263 3.58822263 3.58057678 3.58057678 3.55175343]
Time 3240
[3.77048098 3.71252352 3.71252352 3.68216967 3.68216967 3.66127177
 3.66127177 3.64333974 3.64333974 3.61170673 3.61170673 3.59710873
 3.59710873 3.58822263 3.58822263 3.58057678 3.58057678 3.55175343
 3.55175343 3.17203869]
```

Do you have an idea why this happens?

Best regards

rtimms, 2024-06-05:
Could you also print `step_solution["Time [s]"].entries)`. The solver is probably adding a very small time offset between the final time of the previous step and the first time in the next step (so small that the voltage isn't changing enough to see here).

Tangzl922, 2024-06-05:
The same as the voltages.

```
Time 0
[  0. 360.]
Time 360
[  0. 360. 360. 720.]
Time 720
[   0.  360.  360.  720.  720. 1080.]
Time 1080
[   0.  360.  360.  720.  720. 1080. 1080. 1440.]
Time 1440
[   0.  360.  360.  720.  720. 1080. 1080. 1440. 1440. 1800.]
Time 1800
[   0.  360.  360.  720.  720. 1080. 1080. 1440. 1440. 1800. 1800. 2160.]
Time 2160
[   0.  360.  360.  720.  720. 1080. 1080. 1440. 1440. 1800. 1800. 2160.
 2160. 2520.]
Time 2520
[   0.  360.  360.  720.  720. 1080. 1080. 1440. 1440. 1800. 1800. 2160.
 2160. 2520. 2520. 2880.]
Time 2880
[   0.  360.  360.  720.  720. 1080. 1080. 1440. 1440. 1800. 1800. 2160.
 2160. 2520. 2520. 2880. 2880. 3240.]
Time 3240
[   0.  360.  360.  720.  720. 1080. 1080. 1440. 1440. 1800. 1800. 2160.
 2160. 2520. 2520. 2880. 2880. 3240. 3240. 3600.]
```

rtimms, 2024-06-05:
can you print the differences between times and/or with more precision?

Tangzl922, 2024-06-05:
If you mean a smaller dt, I tried and steps are still repeated.

```
Time 0
[0. 1.]
Time 1
[0. 1. 1. 2.]
Time 2
[0. 1. 1. 2. 2. 3.]
Time 3
[0. 1. 1. 2. 2. 3. 3. 4.]
Time 4
[0. 1. 1. 2. 2. 3. 3. 4. 4. 5.]
Time 5
[0. 1. 1. 2. 2. 3. 3. 4. 4. 5. 5. 6.]
Time 6
[0. 1. 1. 2. 2. 3. 3. 4. 4. 5. 5. 6. 6. 7.]
Time 7
[0. 1. 1. 2. 2. 3. 3. 4. 4. 5. 5. 6. 6. 7. 7. 8.]
Time 8
[0. 1. 1. 2. 2. 3. 3. 4. 4. 5. 5. 6. 6. 7. 7. 8. 8. 9.]
Time 9
[ 0.  1.  1.  2.  2.  3.  3.  4.  4.  5.  5.  6.  6.  7.  7.  8.  8.  9.
  9. 10.]
```

kratman, 2024-07-08:
@MarcBerliner Tagging you since this may be relevant to your work