pybamm-team/PyBaMM #4155: [Bug]: Pybamm does not simulate experiment steps more than 24 hours

aabills, 2024-06-07:
### PyBaMM Version

develop

### Python Version

3.11

### Describe the bug

If a user runs an experiment step that should last more than 24 hours, it only runs for 24 hours. This is expected behavior given https://github.com/pybamm-team/PyBaMM/blob/2cbd788eaf3261c9c3e5001c4436faea363368ce/pybamm/experiment/step/base_step.py#L246, and I recognize that a default max time is probably necessary (not sure if the solvers can take ∞ as the max time 😃), but I think 24 hours is pretty short for a default duration when battery lifetimes are often measured in years and characterization tests (e.g. C/100 pseudo-OCV) that last longer than 24 hours are not unheard of. 

The real issue here is that the code is doing something different than what a reasonable person would expect given a reasonable input, and I think that if a step is terminated due to a non-user specified action such as this, the code should throw a warning or something rather than silently not doing what the user specified. It does so when it hits voltage cutoffs etc.

### Steps to Reproduce

```
import pybamm
model = pybamm.lithium_ion.DFN()
experiment = pybamm.Experiment(["Discharge at C/100 until 2.5 V"])
sim = pybamm.Simulation(model, experiment=experiment)
sol = sim.solve(initial_soc=1)
```

valentinsulzer, 2024-06-11:
Yeah we should definitely change this. The easiest fix is to calculate a better default duration for steps that are constant C-rate or constant current. That should catch most case, and a "rest" should require a duration to avoid division by zero.

kratman, 2024-06-28:
Since there is a larger discussion of this topic in #4224, I am going to close this one for now

closed 2024-06-28 by kratman