pybamm-team/PyBaMM #4156: Tracking Issue for Pytest migration.

prady0t, 2024-06-09:
This is a tracking issue to keep track of all the files that have been successfully migrated to pytest. 

# Integration Tests

* [ ] ├── test_experiments.py 
       ├── `test_models`
* [ ]  │  ├── standard_model_tests.py
* [ ]  │  ├── standard_output_comparison.py
* [ ]  │  ├── standard_output_tests.py
        │  ├── `test_full_battery_models`
        │   │  ├── `test_equivalent_circuit`
* [ ]  │   │   │   └── test_thevenin.py
        │   │  ├── `test_lead_acid`
* [ ]  │   │   │   ├── test_asymptotics_convergence.py
* [ ]  │   │   │   ├── test_compare_basic_models.py
* [ ]  │   │   │   ├── test_compare_outputs.py
* [ ]  │   │   │   ├── test_full.py
* [ ]  │   │   │   ├── test_loqs.py
* [ ]  │   │   │   ├── test_loqs_surface_form.py
        │   │   │  └── `test_side_reactions`
* [ ]  │   │   │       ├── test_full_side_reactions.py
* [ ]  │   │   │      └── test_loqs_side_reactions.py
        │   │  └── `test_lithium_ion`
* [ ]  │   │       ├── base_lithium_ion_half_cell_tests.py
* [ ]  │   │       ├── base_lithium_ion_tests.py
* [ ]  │   │       ├── test_basic_models.py
* [ ]  │   │       ├── test_compare_basic_models.py
* [ ]  │   │       ├── test_compare_outputs.py
* [ ]  │   │       ├── test_compare_outputs_two_phase.py
* [ ]  │   │       ├── test_dfn.py
* [ ]  │   │       ├── test_dfn_half_cell.py
        │   │       ├── `test_external`
* [ ]  │   │        │   └── test_external_temperature.py
* [ ]  │   │       ├── test_initial_soc.py
* [ ]  │   │       ├── test_mpm.py
* [ ]  │   │       ├── test_newman_tobias.py
* [ ]  │   │       ├── test_spm.py
* [ ]  │   │       ├── test_spm_half_cell.py
* [ ]  │   │       ├── test_spme.py
* [ ]  │   │       ├── test_spme_half_cell.py
* [ ]  │   │       ├── test_thermal_models.py
* [ ]  │   │      └── test_yang2017.py
        │ └── `test_submodels`
* [ ]  │       ├── `test_external_circuit`
* [ ]  │        │   └── test_function_control.py
* [ ]  │      └── `test_interface`
* [ ]  │           ├── test_butler_volmer.py
* [ ]  │           ├── test_lead_acid.py
* [ ]  │          └── test_lithium_ion.py
        ├── `test_solvers`
* [ ]  │   ├── test_idaklu.py
* [ ]  │  └── test_solution.py
       └── `test_spatial_methods`
* [ ]     ├── test_finite_volume.py
* [ ]    └── test_spectral_volume.py

# Unit Tests

* [ ] ├── test_batch_study.py
* [ ] ├── test_callbacks.py
* [ ] ├── test_citations.py
       ├── `test_discretisations`
* [ ]  │   └── test_discretisation.py
* [ ] ├── test_doc_utils.py
       ├── `test_experiments`
* [ ]  │   ├── test_experiment.py
* [ ]  │   ├── test_experiment_step_termination.py
* [ ]  │   ├── test_experiment_steps.py
* [ ]  │   └── test_simulation_with_experiment.py
        ├── `test_expression_tree`
* [ ]  │   ├── test_array.py
* [ ]  │   ├── test_averages.py
* [ ]  │   ├── test_binary_operators.py
* [ ]  │   ├── test_broadcasts.py
* [ ]  │   ├── test_concatenations.py
* [ ]  │   ├── test_d_dt.py
* [ ]  │   ├── test_functions.py
* [ ]  │   ├── test_independent_variable.py
* [ ]  │   ├── test_input_parameter.py
* [ ]  │   ├── test_interpolant.py
 * [ ] │   ├── test_matrix.py
        │   ├── `test_operations`
* [ ]  │   │   ├── test_convert_to_casadi.py
* [ ]  │   │   ├── test_copy.py
* [ ]  │   │   ├── test_evaluate_python.py
* [ ]  │   │   ├── test_jac.py
* [ ]  │   │   ├── test_jac_2D.py
* [ ]  │   │   ├── test_latexify.py
* [ ]  │   │   └── test_unpack_symbols.py
* [ ]  │   ├── test_parameter.py
        │   ├── `test_printing`
* [ ]  │   │   ├── test_print_name.py
 * [ ] │   │   └── test_sympy_overrides.py
 * [ ] │   ├── test_scalar.py
 * [ ] │   ├── test_state_vector.py
 * [ ] │   ├── test_symbol.py
* [ ]  │   ├── test_symbolic_diff.py
* [ ]  │   ├── test_unary_operators.py
* [ ]  │   ├── test_variable.py
* [ ]  │   └── test_vector.py
        ├── `test_geometry`
* [ ]  │   └── test_battery_geometry.py
* [ ] ├── test_logger.py
       ├── `test_meshes`
* [ ]  │   ├── test_meshes.py
* [ ]  │   ├── test_one_dimensional_submesh.py
* [ ]  │   ├── test_scikit_fem_submesh.py
* [ ]  │   └── test_zero_dimensional_submesh.py
        ├── `test_models`
* [ ]  │   ├── test_base_model.py
* [ ]  │   ├── test_event.py
        │   ├── `test_full_battery_models`
* [ ]  │   │   ├── test_base_battery_model.py
        │   │   ├── `test_equivalent_circuit`
* [ ]  │   │   │   └── test_thevenin.py
        │   │   ├── `test_lead_acid`
* [ ]  │   │   │   ├── test_base_lead_acid_model.py
* [ ]  │   │   │   ├── test_basic_models.py
* [ ]  │   │   │   ├── test_full.py
* [ ]  │   │   │   └── test_loqs.py
        │   │   └── `test_lithium_ion`
* [ ]  │   │       ├── base_lithium_ion_half_cell_tests.py
* [ ]  │   │       ├── base_lithium_ion_tests.py
* [ ]  │   │       ├── test_Yang2017.py
* [ ]  │   │       ├── test_base_lithium_ion_model.py
* [ ]  │   │       ├── test_basic_models.py
* [ ]  │   │       ├── test_dfn.py
* [ ]  │   │       ├── test_dfn_half_cell.py
* [ ]  │   │       ├── test_electrode_soh.py
* [ ]  │   │       ├── test_mpm.py
* [ ]  │   │       ├── test_mpm_half_cell.py
* [ ]  │   │       ├── test_msmr.py
* [ ]  │   │       ├── test_newman_tobias.py
* [ ]  │   │       ├── test_spm.py
* [ ]  │   │       ├── test_spm_half_cell.py
* [ ]  │   │       ├── test_spme.py
* [ ]  │   │       └── test_spme_half_cell.py
* [ ]  │   ├── test_model_info.py
        │   └── `test_submodels`
* [ ]  │       ├── test_base_submodel.py
* [ ]  │       ├── test_effective_current_collector.py
* [ ]  │       └── test_particle_polynomial_profile.py
        ├── `test_parameters`
* [ ]  │   ├── data_for_testing_2D.csv
* [ ]  │   ├── data_for_testing_3D.csv
* [ ]  │   ├── lico2_diffusivity_Dualfoil1998_2D.json
* [ ]  │   ├── lico2_ocv_example.csv
* [ ]  │   ├── test_base_parameters.py
* [ ]  │   ├── test_bpx.py
* [ ]  │   ├── test_current_functions.py
* [ ]  │   ├── test_ecm_parameters.py
* [ ]  │   ├── test_electrical_parameters.py
* [ ]  │   ├── test_geometric_parameters.py
* [ ]  │   ├── test_lead_acid_parameters.py
* [ ]  │   ├── test_lithium_ion_parameters.py
        │   ├── `test_parameter_sets`
* [ ]  │   │   ├── test_Ai2020.py
* [ ]  │   │   ├── test_Ecker2015.py
* [ ]  │   │   ├── test_Ecker2015_graphite_halfcell.py
* [ ]  │   │   ├── test_LCO_Ramadass2004.py
* [ ]  │   │   ├── test_LGM50_ORegan2022.py
* [ ]  │   │   ├── test_OKane2022.py
* [ ]  │   │   ├── test_OKane2022_negative_halfcell.py
* [ ]  │   │   └── test_parameters_with_default_models.py
* [ ]  │   ├── test_parameter_sets_class.py
* [ ]  │   ├── test_parameter_values.py
* [ ]  │   ├── test_process_parameter_data.py
* [ ]  │   └── test_size_distribution_parameters.py
        ├── `test_plotting`
* [ ]  │   ├── test_plot.py
* [ ]  │   ├── test_plot_summary_variables.py
* [ ]  │   ├── test_plot_thermal_components.py
* [ ]  │   ├── test_plot_voltage_components.py
* [ ]  │   └── test_quick_plot.py
* [ ]  ├── test_pybamm_data.py
        ├── `test_serialisation`
* [ ]  │   └── test_serialisation.py
* [ ]  ├── test_settings.py
* [ ]  ├── test_simulation.py
        ├── `test_solvers`
* [ ]  │   ├── test_algebraic_solver.py
* [ ]  │   ├── test_base_solver.py
* [ ]  │   ├── test_casadi_algebraic_solver.py
* [ ]  │   ├── test_casadi_solver.py
* [ ]  │   ├── test_dummy_solver.py
* [ ]  │   ├── test_idaklu_jax.py
* [ ]  │   ├── test_idaklu_solver.py
* [ ]  │   ├── test_jax_bdf_solver.py
* [ ]  │   ├── test_jax_solver.py
* [ ]  │   ├── test_lrudict.py
* [ ]  │   ├── test_processed_variable.py
* [ ]  │   ├── test_processed_variable_computed.py
* [ ]  │   ├── test_scipy_solver.py
* [ ]  │   └── test_solution.py
        ├── `test_spatial_methods`
* [ ]  │   ├── test_base_spatial_method.py
        │   ├── `test_finite_volume`
* [ ]  │   │   ├── test_extrapolation.py
* [ ]  │   │   ├── test_finite_volume.py
* [ ]  │   │   ├── test_ghost_nodes_and_neumann.py
* [ ]  │   │   ├── test_grad_div_shapes.py
* [ ]  │   │   └── test_integration.py
* [ ]  │   ├── test_scikit_finite_element.py
* [ ]  │   ├── test_spectral_volume.py
* [ ]  │   └── test_zero_dimensional_method.py
* [ ]  ├── test_timer.py
* [ ]  └── test_util.py

kratman, 2024-06-09:
@prady0t Can you summarize what is not covered in #3617 so that we can close the older issue?

agriyakhetarpal, 2024-06-09:
I closed the older issue, just the updates to the Contributing Guide and the removal of the `run-tests.py` file remain, which are being tracked through weekly updates.

prady0t, 2024-09-13:
there are still 14 files left. I should open a new PR concluding everything or should I cover them in pt-7?

agriyakhetarpal, 2024-09-13:
A new PR would be better. Part 6 and Part 7 are ready to merge, I approved them

prady0t, 2024-09-13:
Correction : 9 files

agriyakhetarpal, 2024-09-13:
Almost there with completing them all!

agriyakhetarpal, 2024-09-28:
@prady0t, after #4443, it looks like 8 files are still left:

```bash
❯ rg unittest
```

```bash
tests/testcase.py
4:import unittest
7:class TestCase(unittest.TestCase):

tests/integration/test_models/test_submodels/test_interface/test_butler_volmer.py
8:import unittest
12:class TestButlerVolmer(unittest.TestCase):
387:    unittest.main()

tests/unit/test_expression_tree/test_concatenations.py
5:import unittest.mock as mock

tests/unit/test_parameters/test_parameter_sets/test_Ecker2015.py
6:import unittest
9:class TestEcker2015(unittest.TestCase):
55:    unittest.main()

tests/unit/test_expression_tree/test_functions.py
6:import unittest.mock as mock

tests/unit/test_expression_tree/test_input_parameter.py
7:import unittest.mock as mock

tests/unit/test_expression_tree/test_operations/test_evaluate_python.py
9:import unittest
758:    unittest.main()

tests/integration/test_models/test_full_battery_models/test_lithium_ion/test_compare_outputs.py
7:import unittest
11:class TestCompareOutputs(unittest.TestCase):
153:    unittest.main()
```

Could you migrate these files, too? Also, please check for any additional files that were missed.

prady0t, 2024-09-28:
Thanks for pointing out! Check #4472

kratman, 2024-11-13:
@prady0t Is this done now?

prady0t, 2024-11-15:
Yes, I think this can be closed.

closed 2024-11-15 by prady0t