pybamm-team/PyBaMM #4160: Update Code of Conduct

brosaplanella, 2024-06-10:
PyBaMM's code of conduct is adapted from contributor covenant 1.4. It would be good to update to the newest version (v2.1) as done in https://github.com/pybamm-team/pybamm-cookiecutter/pull/10. 

This probably needs formal approval from the Steering Council at some point, but given than it is an update on our current CoC it should be fine to do it retrospectively (tagging SC members @valentinsulzer @martinjrobins @Scottmar93 @Saransh-cpp @gjo97).

valentinsulzer, 2024-06-10:
Fine by me

gjo97, 2024-06-10:
> This email from ***@***.******@***.***> originates from outside Imperial. Do not click on links and attachments unless you recognise the sender. If you trust the sender, add them to your safe senders list<https://spam.ic.ac.uk/SpamConsole/Senders.aspx> to disable email stamping for this address.

Fine by me too

PR #4171 (2024-06-12): Updated code of conduct to contributor covenant v2.1
# Description

Fixes #4160 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that speeds up the code)
- [ ] Bug fix (non-breaking change which fixes an issue)

# Key checklist:

- [ ] No style issues: `$ pre-commit run` (or  [... cut]
closed 2024-06-12 by kratman