pybamm-team/PyBaMM #4164: SEI + Composite j_sei undefied

kratman, 2024-06-10:
Report from @mpegis:

The PR (#4153) seemed to fix my issue (#4123) when I declared SEI on both Si/Gr phases in Chen2020, but now I am seeing a different error when defining a single SEI on one phase with "none" on the other/cathode. 

```
import pybamm 
model = pybamm.lithium_ion.DFN(
    {
        "particle phases": ("2", "1"),
        "open-circuit potential": (("single", "current sigmoid"), "single"),
        "SEI": (("none","solvent-diffusion limited"),"none")
    }
)
parameters = pybamm.ParameterValues("Chen2020_composite")
experiment = pybamm.Experiment(["Rest for 10 minutes"])
sim = pybamm.Simulation(model, experiment=experiment, parameter_values=parameters)
sol = sim.solve(calc_esoh=False)
```

Is giving me this error

---------------------------------------------------------------------------
UnboundLocalError                         Traceback (most recent call last)
))
--> [182](https://file+.vscode-resource.vscode-cdn.net/Users/mpegis/Coding/voltage_decon_sim/examples/notebooks/~/Coding/voltage_decon_sim/PyBaMM/pybamm/models/submodels/interface/sei/sei_growth.py:182) j_inner = inner_sei_proportion * Arrhenius * j_sei
    [183](https://file+.vscode-resource.vscode-cdn.net/Users/mpegis/Coding/voltage_decon_sim/examples/notebooks/~/Coding/voltage_decon_sim/PyBaMM/pybamm/models/submodels/interface/sei/sei_growth.py:183) j_outer = (1 - inner_sei_proportion) * Arrhenius * j_sei
    [185](https://file+.vscode-resource.vscode-cdn.net/Users/mpegis/Coding/voltage_decon_sim/examples/notebooks/~/Coding/voltage_decon_sim/PyBaMM/pybamm/models/submodels/interface/sei/sei_growth.py:185) variables.update(self._get_standard_concentration_variables(variables))

UnboundLocalError: cannot access local variable 'j_sei' where it is not associated with a value
-----------------------------------------------------------------------------------------------------------

Was curious if I missed another model option or if this is expected behavior?

Thanks!

_Originally posted by @mpegis in https://github.com/pybamm-team/PyBaMM/issues/4123#issuecomment-2159264247_

kratman, 2024-06-10:
@parkec3 Do you want to continue looking into this since you worked on the other PR?

kratman, 2024-06-10:
> @parkec3 Do you want to continue looking into this since you worked on the other PR?

Never mind, @valentinsulzer already made a branch for a fix

mpegis, 2024-06-10:
I will just add that I am trying to explore whats possible for coupled degradation when were using a composite electrode. It's somewhat unclear if I am just making syntax mistakes or if there are limitations to the current approach for separate SEI models on the anode.

So far I have observed a few things

- SEI porosity changes seem to be missing a coupled variable when using SEI on both phases
- A single SEI on one of the phases does not work
- LAM and particle mechanics do not seem to work with a composite electrode - "OptionError: If there are multiple particle phases: 'surface form' cannot be 'false', 'particle size' must be 'single', 'particle' must be 'Fickian diffusion'. Also the following must be 'none': 'particle mechanics', 'loss of active material'"

@valentinsulzer just FYI

PR #4163 (2024-07-18): fix degradation options when one of the phases has no degradation
# Description

Fixes #4164

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that speeds up the code)
- [ ] Bug fix (non-breaking change which fixes an issue)

# Key checklist:

- [ ] No style issues: `$ pre-comm [... cut]
closed 2024-07-18 by kratman