pybamm-team/PyBaMM #4174: [Bug]: Experiencing issues with running the cycling aging experiment example from PyBaMM [Bug]:

wwwy10310, 2024-06-13:
### PyBaMM Version

24.1

### Python Version

3.11.3

### Describe the bug

Here I just run the code for the cycling aging experiment as provided in the PyBaMM example, the link:https://docs.pybamm.org/en/latest/source/examples/notebooks/models/coupled-degradation.html. And the code is wrting as below:
```

```

"However, the processing result indicates an error : [ERROR] callbacks.on_experiment_error(224): Simulation error: Error in Function::call for 'F' [IdasInterface] at .../casadi/core/function.cpp:1432:
Error in Function::call for 'F' [IdasInterface] at .../casadi/core/function.cpp:361:
.../casadi/interfaces/sundials/idas_interface.cpp:596: IDASolve returned "IDA_CONV_FAIL". Consult IDAS documentation.

This error message I have encountered many times. I can list what I have already done to try to fix this error:
1, Playing around conbinations with casadisolver:
solver = pybamm.CasadiSolver(
    mode="safe",
    rtol=1e-6, 1e-5, 1e-4, 1e-3,...
    atol=1e-6, 1e-5, 1e-4, 1e-3,...
    dt_max=100, 10,1, 1e-1, 1e-2, 1e-3,...    
)

2, Adding the some rest time:
[
        (
            "Discharge at 1C until 2.5 V",  # ageing cycles,
            "Rest for 5,15, 30,...minutes",
            "Charge at 0.3C until 4.2 V (5 minute period)",
            "Hold at 4.2 V until C/100 (5 minute period)",
            "Rest for 5,15, 30,... minutes",
        )
    ]* cycle_number

3,  Charging the discharging current with rate variations in experiment design:
"Discharge at 0.1C, 0.2C, 0.5C, 1C, ... until 2.5 V",  # ageing cycles

4, Even I changed the mesh:
var_pts = {
    "x_n": 5, 15, 20 # negative electrode
    "x_s": 5,  15, 20# separator
    "x_p": 5, 15, 20 # positive electrode
    "r_n": 30, 40, 45 # negative particle
    "r_p": 30,  40, 45# positive particle
}

5,  Simulatio mode
SPM: fast, fast with events.
   SPMe: fast fast with events.

Unfortunately, none of above can solve this error, So here, my concern about the parameter set "OKane2022" potentially causing error/issues in the battery cycling aging experiment is invalid. Is it possible that some components or parameters may not be appropriately set for long-term cycling experiments? Is there anyone simulated this Modelling coupled degradation example with long-term secussefuly before? If yes, please give me some advices to modify this code to solve this error, Thank you.

### Steps to Reproduce

```
#1. 
import pybamm
import matplotlib.pyplot as plt
import numpy as np
import pandas as pd

#2
pybamm.set_logging_level("NOTICE")

#3
model = pybamm.lithium_ion.DFN(
    {
    "SEI": "solvent-diffusion limited",
    "SEI porosity change": "true",
    "lithium plating": "partially reversible",
    "lithium plating porosity change": "true" , # alias for "SEI porosity change"
    "particle mechanics": ("swelling and cracking", "swelling only"),
    "SEI on cracks": "true",
    "loss of active material": "stress-driven",
    "calculate discharge energy": "true"
    }
)

# 4 Define the variables
parameter_name = "OKane2022"
parameter_values = pybamm.ParameterValues(parameter_name)
var_pts = {
    "x_n": 5,  # negative electrode
    "x_s": 5,  # separator
    "x_p": 5,  # positive electrode
    "r_n": 30,  # negative particle
    "r_p": 30,  # positive particle
}

#5 Experiment design
cycle_number = 3000

# EV
experiment_EV = pybamm.Experiment(
    [
        "Hold at 4.2 V until C/100 (5 minute period)",
        "Rest for 4 hours (5 minute period)",
        "Discharge at 0.1C until 2.5 V (5 minute period)",  # initial capacity check
        "Charge at 0.3C until 4.2 V (5 minute period)",
        "Hold at 4.2 V until C/100 (5 minute period)",
    ]    
+
[
        (
            "Discharge at 1C until 2.5 V",  # ageing cycles
            "Charge at 0.3C until 4.2 V (5 minute period)",
            "Hold at 4.2 V until C/100 (5 minute period)",
        )
    ]* cycle_number
+
  ["Discharge at 0.1C until 2.5 V (5 minute period)"],  # final capacity check

termination="80% capacity" ,
)

# 6 Sim
sim_EV = pybamm.Simulation(
    model,
    experiment=experiment_EV,
    parameter_values=parameter_values,
    var_pts=var_pts,
  
)

sol_EV = sim_EV.solve(save_at_cycles=[0,1,1000,1500,3000])

capacity_values = sol_EV.summary_variables["Capacity [A.h]"]
sol_EV.save("EV_PYBAMM_example.pkl")

# Print the capacity for the first and last cycles
print(f"EV_PYBAMM_example first cycle: {capacity_values[0]} A.h")
print(f"EV_PYBAMM_example last cycle: {capacity_values[-1]} A.h")
```

### Relevant log output

```shell
2024-06-11 11:38:17.479 - [NOTICE] logger.func(15): Cycle 1920/3006, step 2/3: Charge at 0.3C until 4.2 V (5 minute period)
2024-06-11 11:38:18.484 - [NOTICE] logger.func(15): Cycle 1920/3006, step 3/3: Hold at 4.2 V until C/100 (5 minute period)
2024-06-11 11:38:18.700 - [NOTICE] logger.func(15): Capacity is now 4.835 Ah (originally 5.104 Ah, will stop at 4.083 Ah)
2024-06-11 11:38:18.701 - [NOTICE] logger.func(15): Cycle 1921/3006 (1 hour, 22 minutes, 9 seconds elapsed) --------------------
2024-06-11 11:38:18.701 - [NOTICE] logger.func(15): Cycle 1921/3006, step 1/3: Discharge at 1C until 2.5 V
At t = 290.02 and h = 4.08951e-12, the corrector convergence failed repeatedly or with |h| = hmin.
2024-06-11 11:38:20.050 - [NOTICE] logger.func(15): Cycle 1921/3006, step 2/3: Charge at 0.3C until 4.2 V (5 minute period)
2024-06-11 11:38:21.096 - [NOTICE] logger.func(15): Cycle 1921/3006, step 3/3: Hold at 4.2 V until C/100 (5 minute period)
2024-06-11 11:38:21.315 - [NOTICE] logger.func(15): Capacity is now 4.835 Ah (originally 5.104 Ah, will stop at 4.083 Ah)
2024-06-11 11:38:21.315 - [NOTICE] logger.func(15): Cycle 1922/3006 (1 hour, 22 minutes, 11 seconds elapsed) --------------------
2024-06-11 11:38:21.315 - [NOTICE] logger.func(15): Cycle 1922/3006, step 1/3: Discharge at 1C until 2.5 V
At t = 289.946 and h = 3.85106e-12, the corrector convergence failed repeatedly or with |h| = hmin.
At t = 51.6871 and h = 1.07372e-10, the corrector convergence failed repeatedly or with |h| = hmin.
2024-06-11 11:38:22.672 - [ERROR] callbacks.on_experiment_error(224): Simulation error: Error in Function::call for 'F' [IdasInterface] at .../casadi/core/function.cpp:1432:
Error in Function::call for 'F' [IdasInterface] at .../casadi/core/function.cpp:361:
.../casadi/interfaces/sundials/idas_interface.cpp:596: IDASolve returned "IDA_CONV_FAIL". Consult IDAS documentation.
2024-06-11 11:38:22.673 - [NOTICE] logger.func(15): Capacity is now 4.835 Ah (originally 5.104 Ah, will stop at 4.083 Ah)
2024-06-11 11:38:22.727 - [NOTICE] logger.func(15): Finish experiment simulation, took 1 hour, 22 minutes, 13 seconds
```

agriyakhetarpal, 2024-06-13:
Hello, @wwwy10310 – could you please try running the notebook from the "stable" branch of the documentation, i.e., this link: https://docs.pybamm.org/en/stable/source/examples/notebooks/models/coupled-degradation.html and see if that works for you?

wwwy10310, 2024-06-17:
Hi @agriyakhetarpal. Thank you for your response. I followed your advice and reran the notebook using the original file downloaded from the "stable" branch. However, I encountered the same error:
[image]
Additionally, the plot confirms that the simulation did not complete (it was set for 3000 cycles).
[image]
As you know, even when running the default 10 cycles from the original file, there are always accompanying warnings. I understand that some warnings may not interrupt the simulation process:
[image]
Could you please offer further advice on this error? Thank you.

kratman, 2024-12-23:
Since the original notebook works, and the messages in the last update are just informative (as opposed to errors), I am going to assume this can be closed.

I ran the example provided and I also did not get a convergence failure

closed 2024-12-23 by kratman