pybamm-team/PyBaMM #4183: Compatibility with NumPy v2

agriyakhetarpal, 2024-06-16:
NumPy version 2.0.0 has been released today – we'll need to fix the failing unit tests. The tests are coming from `autograd` and the authors mention that they don't intend to maintain the dependency anymore so we should aim to remove it altogether, either now or later

agriyakhetarpal, 2024-06-16:
As discussed, can be done after the release of v24.5rc0 with an upper pin on NumPy and then removed after v24.5rc1 or with v24.9

brosaplanella, 2024-06-21:
@valentinsulzer shall we reopen this one?

kratman, 2024-11-21:
I have PyBaMM compatible with numpy <=2.0.2, but compatibility with 2.1.0 is still in progress. This is likely needed for upgrading to python 3.13

doelgonzo, 2025-02-06:
Any update on this? I am working within JupyterLite which ships with `numpy==2.0.2`, and of course this is failing to install PyBaMM

kratman, 2025-02-06:
@doelgonzo I actually have a branch working for numpy==2.0.2, but it requires a lot of changes to go much above that. We will likely have to deprecate python 3.9, update Jax, and do a few other changes to get it working. I expect that to be sorted out in a couple months

agriyakhetarpal, 2025-02-06:
> Any update on this? I am working within JupyterLite which ships with `numpy==2.0.2`, and of course this is failing to install PyBaMM

In the meantime, you could use `jupyterlite-pyodide-kernel==0.4.6`, as we shipped Pyodide 0.26.x with that version.

doelgonzo, 2025-04-10:
> > Any update on this? I am working within JupyterLite which ships with `numpy==2.0.2`, and of course this is failing to install PyBaMM
> 
> In the meantime, you could use `jupyterlite-pyodide-kernel==0.4.6`, as we shipped Pyodide 0.26.x with that version.

Unfortunately that results in:
```python
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
Cell In[3], line 1
----> 1 await micropip.install('pybamm', keep_going=True)

File /lib/python3.12/site-packages/micropip/_commands/install.py:146, in install(requirements, keep_going, deps, credentials, pre, index_urls, verbose)
    144 if transaction.failed:
    145     failed_requirements = ", ".join([f"'{req}'" for req in transaction.failed])
--> 146     raise ValueError(
    147         f"Can't find a pure Python 3 wheel for: {failed_requirements}\n"
    148         f"See: {FAQ_URLS['cant_find_wheel']}\n"
    149     )
    151 package_names = [pkg.name for pkg in transaction.pyodide_packages] + [
    152     pkg.name for pkg in transaction.wheels
    153 ]
    155 if package_names:

ValueError: Can't find a pure Python 3 wheel for: 'pybammsolvers', 'casadi==3.6.7'
See: https://pyodide.org/en/stable/usage/faq.html#why-can-t-micropip-find-a-pure-python-wheel-for-a-package
```

agriyakhetarpal, 2025-04-11:
Yes, @doelgonzo, for this, you'll need to either package `pybammsolvers` and restore `casadi` (which I packaged in pyodide/pyodide#4936 but we later disabled it), or switch to a PyBaMM version which does not require `pybammsolvers` as a dependency (i.e., where the IDAKLU solver is optional); this would be v24.11, I think. 

https://github.com/agriyakhetarpal/pybamm-lite/ is the repository where I've been trying this out (see also #3910) – I've been using a custom Pyodide distribution there, after which the changes could be upstreamed once done.

PR #4196 (2024-06-21): #4183 remove autograd
# Description

Removes autograd dependency and thus enables update to numpy 2.0.
The "differentiate numpy function directly" functionality, which relied on autograd, is deprecated, but I am not aware of any cases where it was being used, since we switched to implementing our own functions and specifying their derivatives.

Fixes #4183 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to  [... cut]
closed 2025-09-18 by MarcBerliner