pybamm-team/PyBaMM #4200: Applying arbitrary temperature and voltage profiles in an experiment step

valentinsulzer, 2024-06-19:
### Discussed in https://github.com/pybamm-team/PyBaMM/discussions/4173

<div type='discussions-op-text'>

<sup>Originally posted by **jhthompson12** June 12, 2024</sup>
Hello, 

I have a few years of operational cell data, specifically cell voltage and ambient temperature, that I would like to use as the drive cycle input for a PyBaMM simulation via `pybamm.step.voltage`

Im breaking my simulation into daily experiment cycles where I create a `pybamm.step.voltage(drive_cycle)` step for each day with the time series voltage data for that day. Is there a way to also include the timeseries ambient temperature data as an input to the experiment? 

I see that there is a [`temperature` kwarg](https://docs.pybamm.org/en/stable/source/api/experiment/experiment_steps.html#pybamm.step._Step) that can be provided to `pybamm.step.voltage`, but it looks like this can only be a fixed value? 

> temperature ([float](https://docs.python.org/3/library/functions.html#float) or string, optional) – The temperature of the step. If a float, the value is in Kelvin. If a string, the value should be a valid temperature string, e.g. “25 oC”.temperature ([float](https://docs.python.org/3/library/functions.html#float) or string, optional) – The temperature of the step. If a float, the value is in Kelvin. If a string, the value should be a valid temperature string, e.g. “25 oC”.temperature ([float](https://docs.python.org/3/library/functions.html#float) or string, optional) – The temperature of the step. If a float, the value is in Kelvin. If a string, the value should be a valid temperature string, e.g. “25 oC”.

Is it possible to define this as a timeseries like I am able to do with the voltage profile that I provide to `pybamm.step.voltage` as a numpy array with a time and voltage column? </div>

Rishab87, 2025-02-16:
Hi, I would like to work on this issue, can it be assigned to me?

Rishab87, 2025-02-17:
I've created a PR for it, can you please review it and tell if anything's wrong?