pybamm-team/PyBaMM #4202: [Bug]: I have a problem with pybamm.ParameterValues("Prada2013")

ErfanSamadi1998, 2024-06-20:
### PyBaMM Version

24.1

### Python Version

3.11.6

### Describe the bug

1.
when I use Prada2013 for ParameterValues in pybam. like this line:
parameter_values = pybamm.ParameterValues("Prada2013")

then when I want to add options to model like this line:
model = pybamm.lithium_ion.SPM(options=options)

I can not use thermal option for Prada2013. cause i see this error:
KeyError: "'Negative current collector thickness [m]' not found. Best matches are ['Negative electrode thickness [m]', 'Positive electrode thickness [m]', 'Separator thickness [m]']"

it seems not relative.

2.
and there is a second problem with Prada2013. when you want to run this line:
sim.plot(["Current [A]"])

you will see that the current is multiplied by -1 
this problem will not appear when you use for example chen2020 instead of Prada2013

### Steps to Reproduce

1.
options = {"thermal": "lumped"}
model = pybamm.lithium_ion.SPM(options=options)
parameter_values = pybamm.ParameterValues("Prada2013")
experiment = pybamm.Experiment(
    [
        (
            "Charge at 4C until 3.6 V",
            "Hold at 3.6 V until C/20",
            "Discharge at 4C until 2V",
            "Hold at 2 V until C/20"
            
            
        ),
    ]
    * 5
)

sim = pybamm.Simulation(model, parameter_values=parameter_values, experiment=experiment)
sim.solve(initial_soc=0.5)

2.
model = pybamm.lithium_ion.SPM()
parameter_values = pybamm.ParameterValues("Prada2013")
experiment = pybamm.Experiment(
    [
        (
            "Charge at 4C until 3.6 V",
            "Hold at 3.6 V until C/20",
            "Discharge at 4C until 2V",
            "Hold at 2 V until C/20"
        ),
    ]
    * 5
)

sim = pybamm.Simulation(model, parameter_values=parameter_values, experiment=experiment)
sim.solve(initial_soc=0.5)
sim.plot(["Current [A]"])

kratman, 2024-06-24:
@ErfanSamadi1998 I can confirm that neither Prada2013 or Xu2019 have those parameters. We probably need to go into the original papers to see if these values are known.

As a temporary solution you can copy values from a similar cell.

valentinsulzer, 2024-06-25:
> you will see that the current is multiplied by -1

PyBaMM follows the "Newman" standard with positive current for discharge and negative current for charge

brosaplanella, 2024-06-28:
> @ErfanSamadi1998 I can confirm that neither Prada2013 or Xu2019 have those parameters. We probably need to go into the original papers to see if these values are known.
> 
> As a temporary solution you can copy values from a similar cell.

Yes, Prada2013 does not provide thermal parameters. If you want to use a thermal model you will need to add the parameters manually as Eric suggested.