pybamm-team/PyBaMM #4206: Use `tempfile` for temporary JSON files.

prady0t, 2024-06-23:
After merging #4125 , we generate a lot of temporary JSON files while running tests. These files are short-lived and are created and deleted with every test passing, although if the tests fail these files have to be deleted manually.

You can see the number of json files being generated when test suit fails many times : 

[image]

We can use `tempfile` to generate temporary files to fix this. See : https://docs.python.org/3/library/tempfile.html

prady0t, 2024-06-23:
You can assign this to me, I will open a PR soon.

kratman, 2024-06-24:
This is true for notebooks as well. They make images and other files, but do not clean them up

agriyakhetarpal, 2024-06-24:
For the notebooks, I think it's better to not use tempfiles so as to not mislead newcomers to PyBaMM and Python, plus users might want to inspect the images after the notebook/cell has been executed

PR #4270 (2024-07-16): Using `tempfile` for `standard_model_test`
# Description

Tempfile avoids race conditions securely hence we can remove `uuid` to manually create random file names.

Fixes #4206
closed 2024-07-16 by agriyakhetarpal