pybamm-team/PyBaMM #4218: [Bug]: latexify issue showing lithium plating equations

pkalbhor, 2024-06-26:
### PyBaMM Version

24.1

### Python Version

3.10.13

### Describe the bug

Gives error when try to display model with lithium plating option.
```bash
KeyError: 'min'
```

### Steps to Reproduce

```python
import pybamm
model = pybamm.lithium_ion.SPM(
                            options={
                                "lithium plating": "irreversible"
                                }
                            )
model.latexify()
```

Manimaran-tech, 2025-08-05:
Hi! 👋
I'm new to PyBaMM and interested in contributing. I’d love to take up this issue and help resolve the LaTeX formatting for the lithium plating equations. Could you please assign this to me if it's still available?

agriyakhetarpal, 2025-08-08:
Done, @Manimaran-tech. Thanks for taking this on!

swastim01, 2025-10-28:
Hi! I noticed this issue was still open and decided to look into it. I’ve implemented a fix and opened PR #5245, I'm open to any feedback or suggestions. Thanks.