pybamm-team/PyBaMM #4223: remove python-idaklu

martinjrobins, 2024-06-27:
### Description

The python-idaklu solver is a legacy solver that uses idaklu but calls back into python for the functions. With the recent improvements to idaklu this can now be removed.

### Motivation

reduce the number of solvers we need to maintain

kratman, 2024-06-28:
@martinjrobins It would be nice to break out idaklu into its own package within the PyBaMM ecosystem. That would let PyBaMM just install it as a library similar to how we install a casadi package. Is it feasible to do that along with removing the old solver?

agriyakhetarpal, 2024-06-28:
>  It would be nice to break out idaklu into its own package within the PyBaMM ecosystem

Note for other readers: this was discussed in the monthly developer meeting in April and in the recent v24.5 release prep meeting. @Saransh-cpp reported that this could be a suitable 175-hour GSoC project

rtimms, 2024-07-06:
@MarcBerliner not sure if you've seen this but I know you were looking at IDAKLU

PR #4326 (2024-08-27): Remove python-idaklu
# Description

Removes the legacy python-idaklu solver which calls IDAKLU but returns back to python for certain functions. With recent improvements to IDAKLU this can now be removed.

Reduces the number of solvers we need to maintain.

Fixes #4223 

## Type of change

- [x] Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [x] Optimization (back-end change that speeds up t [... cut]