pybamm-team/PyBaMM #4229: [Bug]: Why does the current not change over time when doing a drive cycle?

ADSDAW, 2024-06-30:
### PyBaMM Version

22.9

### Python Version

3.7.0

### Describe the bug

The code and graphics are as follows
US06.csv is copied from the pybamm directory
[image]

### Steps to Reproduce

[image]

kratman, 2024-07-01:
@ADSDAW As I mentioned in #4233, you are using a rather old version of PyBaMM. Can you try again with a newer version?

With the following code (and the pre-release of PyBaMM 24.5) I do not get the behavior you are reporting. This is mostly what you did, except I used the data loader class since the drive cycles are no longer stored directly in PyBaMM.
```python
import pybamm
import pandas as pd
file_path = pybamm.DataLoader().get_data("US06.csv")
parameter_values = pybamm.ParameterValues("Chen2020")
drive_cycle = pd.read_csv(file_path, comment="#", header=None).to_numpy()
current_interpolant = pybamm.Interpolant(drive_cycle[:, 0], drive_cycle[:, 1], pybamm.t)
parameter_values["Current function [A]"] = current_interpolant
model = pybamm.lithium_ion.DFN()
sim = pybamm.Simulation(model, parameter_values=parameter_values)
sol = sim.solve()
sim.plot(["Current [A]", "Voltage [V]"])
```
[image]

ADSDAW, 2024-07-02:
Thank  u vary much

kratman, 2024-07-02:
I will close this issue for now since I suspect it was mainly caused by the old python version.

closed 2024-07-02 by kratman