pybamm-team/PyBaMM #4266: Rename `set_events`

MehrdadBabazadeh, 2024-07-15:
This adds a side effect to what would be expected to be a simple setter. At minimum it should be changed to `append_events()` or `add_events()`. Was there an actual bug that this was fixing or just not sure if it would cause issues?

_Originally posted by @kratman in https://github.com/pybamm-team/PyBaMM/pull/4254#discussion_r1672332716_

brosaplanella, 2024-07-15:
I would also vote for using a dictionary instead of a list, so it is easier to access a specific event in case the user wants to modify in.

kratman, 2024-07-15:
> I would also vote for using a dictionary instead of a list, so it is easier to access a specific event in case the user wants to modify in.

The one advantage of a list over a dictionary is the ordering in a list. So if we are appending to a list then the events are in some order. We can use an "OrderedDict" object to get both

valentinsulzer, 2024-07-25:
Ordering is not super important, and Python's dict is an `OrderedDict` since Python 3.6

medha-14, 2024-09-06:
The changes are to be made in `ocv_element.py `and `diffusion_element.py`  or only one of them as I see the issue to be that the `set_events` method is appending the  events rather than  replacing it.
Shall I make a PR for the same?

agriyakhetarpal, 2024-09-06:
Hi @medha-14, let's wait for a few days for a response from @MehrdadBabazadeh since they are assigned to this issue. Otherwise, please feel free to take it up!

kratman, 2024-09-06:
@medha-14, @agriyakhetarpal This ticket has been open for a while, and can safely be taken over.

@medha-14 The core issue is that all versions of `set_events` are not actually setting the events, but rather appending to the existing list. The smallest change for this would be to rename the function to something like `add_events_from` so that the code would clearly read along the lines of `model.add_events_from(new_events)`. This makes it far more clear what the function is doing, but keep in mind that `set_events` is not used only in the `ocv_element.py` and `diffusion_element.py` files. You need to change ALL of the usages, not just the ones in those two files.

I think things like switching it to a dictionary can come afterwards. Lets get the rename done first since that does not require too many other changes

medha-14, 2024-09-06:
Thank you for the response, I have made the necessary changes and put them up for review.

PR #4421 (2024-09-06): Renames `set_events` to `add_events_from`
# Description

I have renamed `set_events` to `add_events_from`  in all occurrences

Fixes #4266 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that speeds up the code)
- [ ] Bug fix (non-breaking change which fixes an issue)

# Key [... cut]
closed 2024-09-06 by kratman