pybamm-team/PyBaMM #4274: Clarify definition of PyBaMM's SPMe

brosaplanella, 2024-07-17:
The definition of the SPMe remains a sticking point in PyBaMM. Even though the cited article for its definition is the Marquis et al (2019) paper, the actual implementation is a bit different (more accurate) as some terms have been "unlinearised"*. As part of improving the Creating Models notebooks (#3844) I wanted to write a BasicSPMe file and document the SPMe definition better. However, I have also noticed that the current approach falls somewhat in the middle: some terms are unlinearised and some are not. The marginal cost of doing the SPMe fully "unlinearised". All this discussion pertains to `composite_conductivity.py`

*by "unlinearised" I mean that some terms in the asymptotic expansion get later grouped again in their nonlinear form.

## Current implementation

[image]

$$
\phi_\mathrm{e,const} = - \overline{\Delta \phi}_\mathrm{n} + \overline{\phi}_\mathrm{n}  - \left[ \chi(\overline{c}_\mathrm{e}) \frac{R T}{F} \frac{1}{L_\mathrm{n}} \int_0^{L_\mathrm{n}} \log \left(\frac{c_\mathrm{e,n}}{\overline c_\mathrm{e}} \right) \mathrm{d} x - i_\mathrm{app} L_\mathrm{n} \left(\frac{1}{3 \kappa_\mathrm{e,eff} (\overline{c}_\mathrm{e,n})} - \frac{1}{\kappa_\mathrm{e,eff} (\overline{c}_\mathrm{e,s})} \right)\right]
$$

[image]

$$
\phi_\mathrm{e,n} = \phi_\mathrm{e,const}  + \chi(\overline{c}_\mathrm{e,n}) \frac{R T}{F} \log \left( \frac{c_\mathrm{e,n}}{\overline{c}_\mathrm{e}} \right) - \frac{i_\mathrm{app}}{\kappa_\mathrm{e,eff} (\overline{c}_\mathrm{e,n})} \frac{x_\mathrm{n}^2 - L_\mathrm{n}^2}{2 L_\mathrm{n}} - \frac{i_\mathrm{app} L_\mathrm{n}}{\kappa_\mathrm{e,eff} (\overline{c}_\mathrm{e,s})}
$$

[image]

$$
\phi_\mathrm{e,s} = \phi_\mathrm{e,const}  + \chi(\overline{c}_\mathrm{e,s}) \frac{R T}{F} \log \left( \frac{c_\mathrm{e,s}}{\overline{c}_\mathrm{e}} \right) - - \frac{i_\mathrm{app} x_\mathrm{s}}{\kappa_\mathrm{e,eff} (\overline{c}_\mathrm{e,s})}
$$

[image]

$$
\phi_\mathrm{e,p} = \phi_\mathrm{e,const}  + \chi(\overline{c}_\mathrm{e,p}) \frac{R T}{F} \log \left( \frac{c_\mathrm{e,p}}{\overline{c}_\mathrm{e}} \right) - - \frac{i_\mathrm{app} }{\kappa_\mathrm{e,eff} (\overline{c}_\mathrm{e,p})} \frac{x_\mathrm{p} (2 L_x - x_\mathrm{p}) + L_\mathrm{p}^2 - L_x^2}{2 L_\mathrm{p}}
$$

## Proposed implementation
(Work in progress, will finish this tomorrow)

valentinsulzer, 2024-07-18:
Sounds good to me

rtimms, 2024-07-26:
Me too, thanks!

ejfdickinson, 2025-07-16:
@rtimms @brosaplanella Is there now a quick win here by documenting that PyBaMM implements SPMe according to BPX v1.0? Are there any features that we wrote in for BPX where PyBaMM might deviate?

ejfdickinson, 2025-08-28:
@brosaplanella The BPX v1.0 definitions are now public, so I suggest you check directly with the latest document (n.b. bugs as noted above), @rtimms has a foot in both camps and has drafted this, so I think he'll be able to give a better/faster answer than I can, but I think the presumption is that PyBaMM compliance with the standard should require minimal effort at the PyBaMM side!