pybamm-team/PyBaMM #4277: Pipelining parameter sweeping for long-term degradation modelling

RuiheLi, 2024-07-18:
### Description

This relates to a tool ParaSweeper proposed in this [paper](https://www.sciencedirect.com/science/article/pii/S0378775324001356#da0010), which is mainly about pipelining parameter sweeping for long-term degradation modelling. It aims to generate large amount of long-term degradation modelling data, which can later be used for sensitivity analysis and optimization. It might be a nice feature to add on PyBaMM.

### Motivation

I notice there are questions about degradation modelling on [PyBaMM Discussions](https://github.com/pybamm-team/PyBaMM/discussions?discussions_q=degradation) from time to time involving how to fit the model to ageing datasets and deal with multiple errors. This tool ParaSweeper may help a lot on those questions. 
There are three useful features of ParaSweeper:

1. Can be tailored for HPC system
2. Run and check – always return something and not always assume the simulation runs well (all other tools assume the model runs well)
3. Tidy up input/outputs 

For these reasons, I think it can be a upgrade for the [BatchStudy](https://docs.pybamm.org/en/stable/source/examples/notebooks/batch_study.html) or [Degradation experiments with reference performance tests example](https://docs.pybamm.org/en/stable/source/examples/notebooks/simulations_and_experiments/rpt-experiment.html). 

### Possible Implementation

ParaSweeper has already been [here](https://github.com/RuiheLi/PyBaMM/tree/Li_et_HPC_paper/scripts/HPC_Li_et_al/ParaSweeper). But I am happy to re-write it, break into little chunks and do whatever you may thinks that can become more compatible with PyBaMM. Any suggestions will be appreciated.

valentinsulzer, 2024-07-18:
Open to this if it can be sufficiently generalized, but in general the PyBaMM development philosophy should be to be as model-agnostic as possible and work on things that make all models more robust. Opinionated functions that do very specific things for specific models are valuable but should live outside of the main PyBaMM repository.

kratman, 2024-08-27:
I am going to close this for now. We can re-open it as a discussion if needed

closed 2024-08-27 by kratman