pybamm-team/PyBaMM #4280: [Bug]:  the corrector convergence failed repeatedly or with |h| = hmin.

simdieudoboinzemwende, 2024-07-20:
### PyBaMM Version

24.1

### Python Version

3.12.3

### Describe the bug

Hello, I am a PhD student working on a hybrid modeling project for lithium-ion batteries, integrating neural networks with the SPM model. I would like to know if anyone here has worked on a similar topic.
I am also encountering some issues with my simulation experiments. Specifically, I am unable to perform high-rate experiments with the DFN model. For example, starting from 3C, I encounter convergence problems. I have tried changing the solver and modifying the hyperparameters, but the error persists.

### Steps to Reproduce

model = pybamm.lithium_ion.DFN()
params = pybamm.ParameterValues('Chen2020')
experiment = pybamm.Experiment([
    "Discharge at 1C for 1 hours",
    "Rest for 10 minutes",
    "Charge at 3C until 4.2V",
    "Hold at 4.2V until C/50",
])
sim = pybamm.Simulation(model, parameter_values=params, experiment=experiment)
sim.solve(solver=pybamm.CasadiSolver(mode="safe", dt_max=1e-5))
sim.plot()

### Relevant log output

```shell
2024-07-20 19:36:15.190 - [NOTICE] logger.func(15): Cycle 1/4 (834.800 us elapsed) --------------------
2024-07-20 19:36:15.193 - [NOTICE] logger.func(15): Cycle 1/4, step 1/1: Discharge at 1C for 1 hours
2024-07-20 19:36:16.040 - [NOTICE] logger.func(15): Cycle 2/4 (851.187 ms elapsed) --------------------
2024-07-20 19:36:16.041 - [NOTICE] logger.func(15): Cycle 2/4, step 1/1: Rest for 10 minutes
2024-07-20 19:36:16.407 - [NOTICE] logger.func(15): Cycle 3/4 (1.218 s elapsed) --------------------
2024-07-20 19:36:16.408 - [NOTICE] logger.func(15): Cycle 3/4, step 1/1: Charge at 4C until 4.1V
At t = 192.007 and h = 8.10334e-09, the corrector convergence failed repeatedly or with |h| = hmin.
At t = 191.949 and h = 3.0101e-09, the corrector convergence failed repeatedly or with |h| = hmin.
2024-07-20 19:36:19.510 - [NOTICE] logger.func(15): Cycle 4/4 (4.321 s elapsed) --------------------
2024-07-20 19:36:19.511 - [NOTICE] logger.func(15): Cycle 4/4, step 1/1: Hold at 4.2V until C/50
2024-07-20 19:36:21.000 - [NOTICE] logger.func(15): Finish experiment simulation, took 5.809 s
```

kratman, 2024-07-22:
As far as the message:
```
... the corrector convergence failed repeatedly or with |h| = hmin.
```
This is fairly normal. It is a logging message that comes out of the solver as it is dynamically adjusting the timestep. Is your simulation actually failing or are you just seeing that message in the logs?

Generally adjusting the timestep and/or number of grid points in the discretization would be the approach for solving convergence issues. There is some work being done to improve the solver performance at higher C-rates, but these have not been merged yet. Unfortunately, some of the parameter sets struggle at higher C rates, so you might have difficulties with some of the simulations.

simdieudoboinzemwende, 2024-07-23:
Alright, thank you for your response. Actually, the convergence problem mentioned, if I go to 4C or 5C, repeats recursively until the simulation fails. However, I am currently conducting a study where I need data up to at least 5C from the DFN model for my training system.

Again, thank you very much for your attention. If you would like to know more about my modeling project, I would be happy to share it with you.

MarcBerliner, 2024-07-23:
@simdieudoboinzemwende we are actively working on improving simulation robustness using the `IDAKLUSolver`. These changes will soon be merged into the `develop` branch of PyBaMM and will be in the official 24.9 release this Fall. With these upcoming changes, we can successfully charge at 5C using the improved IDAKLU solver.
```python
import pybamm

model = pybamm.lithium_ion.DFN()
params = pybamm.ParameterValues('Chen2020')
experiment = pybamm.Experiment([
"Discharge at 1C for 1 hours",
"Rest for 10 minutes",
"Charge at 5C until 4.2V",
"Hold at 4.2V until C/50",
])
sim = pybamm.Simulation(model, parameter_values=params, experiment=experiment)
sol = sim.solve(solver=pybamm.IDAKLUSolver())
sim.plot()
```
[image]

simdieudoboinzemwende, 2024-07-23:
Thank you for your contributions.

kratman, 2024-07-24:
@simdieudoboinzemwende the changes from @MarcBerliner's work have been merged. If you use the develop branch of PyBaMM, then your simulations would work a bit better. You just need to use the IDAKLU solver.

simdieudoboinzemwende, 2024-08-05:
Okay, great! PyBaMM has an amazing team. I would be thrilled to join such a team. Thank you.

> Les modifications de l’œuvre de ont été fusionnées. Si vous utilisez la branche develop de PyBaMM, vos simulations fonctionneront un peu mieux. Il vous suffit d’utiliser le solveur IDAKLU.

closed 2024-08-05 by kratman

simdieudoboinzemwende, 2024-08-07:
Hello, I followed the steps for installing IDAKLUSOLVER. Since then, I have been trying to update to version 24.9 as you suggested, but I can’t manage to do it. The latest version I can get is 24.5.

I am using conda and this is the code I enter: “pip install pybamm --upgrade”

agriyakhetarpal, 2024-08-07:
Hi @simdieudoboinzemwende, PyBaMM version 24.9 is not available yet – you will need to build the IDAKLU solver from source. Here are the instructions for doing so: https://docs.pybamm.org/en/latest/source/user_guide/installation/index.html#full-installation-guide

kratman, 2024-08-07:
@simdieudoboinzemwende PyBaMM has official releases every 4 months or so. 24.5 is the May 2024 release (which was delayed a bit) and 24.9 is the release planned for September 1st 2024.

In the meantime, building from source with the development branch as @agriyakhetarpal suggested is the best option.

nuanyuanyuanyuan, 2025-02-26:
> Hi [@simdieudoboinzemwende](https://github.com/simdieudoboinzemwende), PyBaMM version 24.9 is not available yet – you will need to build the IDAKLU solver from source. Here are the instructions for [...]

i am struggling with the same question. And i don't know how to build IDAKLU solver from the instructions you mentiond. would you please give more details on this? thanks a lot.

nuanyuanyuanyuan, 2025-02-26:
> Hi [@simdieudoboinzemwende](https://github.com/simdieudoboinzemwende), PyBaMM version 24.9 is not available yet – you will need to build the IDAKLU solver from source. Here are the instructions for [...]

Hi, i just found that my version has been 24.9.0 already, but the problem is still there.

MarcBerliner, 2025-02-26:
Hi @nuanyuanyuanyuan, the latest pybamm version 25.1.1 installs the `IDAKLUSolver` by default. Please update to the latest version and try again.

nuanyuanyuanyuan, 2025-03-01:
> Hi [@nuanyuanyuanyuan](https://github.com/nuanyuanyuanyuan), the latest pybamm version 25.1.1 installs the `IDAKLUSolver` by default. Please update to the latest version and try again.

i have upgraded to version 25.1.1. However, the problem is not fixed. I guess it is due to the initial parameter setting, but i have no ideal which parameter is wrong. Do you have any suggestion?

MarcBerliner, 2025-03-03:
@nuanyuanyuanyuan if you have a minimum working example of the code that is not working, please make a post to https://pybamm.discourse.group/