pybamm-team/PyBaMM #4292: [Bug]: Bug with the calculation of PE porosity in porosity sub-model

srawat77, 2024-07-24:
### PyBaMM Version

Latest version

### Python Version

3.10.12

### Describe the bug

When I was using a combination of Prada + OKane'23 + Other literature parameters for the degradation modeling of LFP cells. The model options were:

'Model option': "{'SEI': 'interstitial-diffusion limited', 'SEI on cracks': 'true', 'lithium plating': 'none', 'particle mechanics': 'constant cracks', 'loss of active material': 'none', 'contact resistance': 'true', 'open-circuit potential': 'current sigmoid', 'SEI film resistance': 'distributed', 'SEI porosity change': 'true', 'thermal': 'lumped'}",

When the simulation was run, PE porosity in the first cycle itself (of degradation modeling) was updated by a constant factor (For Example If I input PE porosity =0.3, the updated porosity in the first cycle was magically changed to 0.6262776 i.e. a factor of 0.3262776). 

I realized that reaction_driven_porosity sub-model (path: pybamm/models/submodels/porosity/reaction_driven_porosity.py) is updating PE_porosity by a constant factor of 0.3262776 and the value of this constant factor is dependent upon the choice of user inputs for PE such as lithium plating thickness, dead lithium thickness, total SEI on cracks thickness, electrode roughness ratio. If the user is not mindful of the choice of inputs, the parameter 'delta_eps_k' can be positive leading to an increase in PE porosity in the first cycle itself. 

[image]

### Steps to Reproduce

Refer the image and description of the bug

valentinsulzer, 2024-07-24:
That's strange, `L_tot` should be zero initially since we subtract the initial SEI thicknesses and the initial plating and dead lithium thicknesses are zero. Can you plot each thickness vs time?

srawat77, 2024-07-24:
Thank you for the prompt response, yes I will plot the thicknesses and post them here.

closed 2024-08-24 by github-actions[bot]