pybamm-team/PyBaMM #4332: [Bug]: hysteresis state error

llh929423719, 2024-08-11:
### PyBaMM Version

24.5

### Python Version

3.9

### Describe the bug

Under some specific parameter sets, hysteresis state will exceed the range of (-1,1)，adding bounds in WyciskOpenCircuitPotential submodel seems to have no effect.
[image]

### Steps to Reproduce

 my hysteresis paramset:
'Secondary: Negative particle hysteresis decay rate': 0.005,
'Secondary: Negative particle hysteresis switching factor': 0.2,

rtimms, 2024-08-11:
@mleot

mleot, 2024-08-12:
@llh929423719 can you provide more details on the specific base parameter set, the experimental protocol, the model used, and the solver? I would like to try and replicate the error you are experiencing.

llh929423719, 2024-08-15:
paramset:

import pybamm
import os
import numpy as np

y_low = 0.18
y_up = 0.819
y_mid = (y_low + y_up) / 2

x_gr_low = 0.053410
x_gr_up = 0.86455
x_gr_mid = (x_gr_low + x_gr_up) / 2
x_si_low = 0.18072
x_si_up = 0.82146
x_si_mid = (x_si_low + x_si_up) / 2

def si_lith_yk(sto):
    si_lith_sto = np.array([0.000446504811095948, 0.00563431441023561, 0.0181660574920546,
                            0.0334360461827143, 0.0460270060067983, 0.0613297753940689,
                            0.0784344256531157, 0.108160701549205, 0.135206891136893,
                            0.185718772288891, 0.250676366228057, 0.292182015346653,
                            0.340003235771988, 0.378801305094702, 0.404968231803938,
                            0.445567124556602, 0.485262433268812, 0.525858153695997,
                            0.565552404966381, 0.606152355160871, 0.651267053232150,
                            0.698186804500684, 0.744206144054243, 0.791126952764602,
                            0.834437655080453, 0.852483957285695, ])

    si_lith_ocp = np.array([0.994727595054768, 0.742820630901517, 0.627991965557005,
                            0.547139434807918, 0.497922919892966, 0.453391400988814,
                            0.405343888012681, 0.342057580879760, 0.309235644037588,
                            0.276399960451676, 0.249414045048080, 0.237673268452228,
                            0.223585499723215, 0.211846309290103, 0.204801103123313,
                            0.188374273074835, 0.170776326203885, 0.150834559525251,
                            0.132064967110916, 0.116809782605822, 0.103895245582934,
                            0.0909796511182185, 0.0804078764611865, 0.0686639275398565,
                            0.0569220935021789, 0.0522249369103773,
                            ])

    return pybamm.Interpolant(x=si_lith_sto, y=si_lith_ocp,
                              children=sto, interpolator='linear', )

# si_lith_yk.__name__ = 'silicon_ocp_lithiation_yk'
def si_delith_yk(sto, ):
    si_delith_sto = np.array([0.000443332485617468, 0.00853329117665252, 0.0220278350411900,
                              0.0364249055043548, 0.0535263834379232, 0.0724413741033554,
                              0.0940650020064959, 0.113889921363339, 0.138224301387840,
                              0.165268376091875, 0.194118561434991, 0.225679498899467,
                              0.256341082090795, 0.293325638681645, 0.328506199517066,
                              0.368198335903798, 0.406084361651449, 0.440361338446417,
                              0.474634085474080, 0.509808301658545, 0.557614717898313,
                              0.600915903237728, 0.640612269391765, 0.684821268538938,
                              0.726322687890229, 0.765114412561986, 0.816524590545307,
                              0.839051802488901, 0.849834536790252, 0.853386483884322,
                              ])

    si_delith_ocp = np.array([0.991212658424613, 0.954886888091460, 0.906841489998980, 0.858795563185586,
                              0.807233113579298, 0.765042770878264, 0.724022487557875, 0.690033134939623,
                              0.652526202086650, 0.617360974157708, 0.583366334330325, 0.552885044970358,
                              0.525919220961458, 0.504807923623092, 0.484869329269937, 0.463756445768832,
                              0.441472974166168, 0.420363262990541, 0.394566969641374, 0.367598502027909,
                              0.337107695691506, 0.314821051763364, 0.298394750435798, 0.281965805503668,
                              0.265538446734276, 0.246769383040854, 0.209246588560489, 0.169397422062572,
                              0.116667027959291, 0.0522244081894643,
                              ])

    return pybamm.Interpolant(x=si_delith_sto, y=si_delith_ocp,
                              children=sto, interpolator='linear', )

# si_delith_yk.__name__ = 'silicon_ocp_delithiation_yk'
[... 849 lines cut ...]
        "Volume-averaged cell temperature [C]",
        # "Terminal voltage [V]",
        # "Negative lithium plating thickness [m]",
        # "R-averaged negative particle concentration [mol.m-3]",
        # "Negative total SEI thickness [m]",
        "Negative primary particle concentration [mol.m-3]",
        "Negative secondary particle concentration [mol.m-3]",
        "Positive particle concentration [mol.m-3]",
        "Current [A]",
        "Electrolyte concentration [mol.m-3]",
        "Terminal voltage [V]",
        "Negative electrode secondary hysteresis state",
        "X-averaged negative electrode secondary hysteresis state",
        "Negative electrode secondary hysteresis state distribution",
        "Sum of x-averaged negative electrode primary volumetric interfacial current densities [A.m-3]",
        "Sum of x-averaged negative electrode secondary volumetric interfacial current densities [A.m-3]",
        # "Negative electrode primary exchange current density [A.m-2]",
        # "Negative electrode secondary exchange current density [A.m-2]",
        # "Negative electrode potential [V]",
        # "Negative electrolyte potential [V]",
    ],
    time_unit="seconds",
    spatial_unit="um",
)
plot.dynamic_plot()

llh929423719, 2024-08-15:
After reading the literature, I wondered if it was due to the current problem. I modified the current calculation method and it seemed to be improved. The following is my modified wycisk_ocp(mainly modified set_rhs):
```

#
# from Wycisk 2022
#
import pybamm
from . import BaseOpenCircuitPotential

class WyciskOpenCircuitPotential(BaseOpenCircuitPotential):
    """
    Class for open-circuit potential with hysteresis based on the approach outlined by Wycisk :footcite:t:'Wycisk2022'.
    This approach employs a differential capacity hysteresis state variable. The decay and switching of the hysteresis state
    is tunable via two additional parameters. The hysteresis state is updated based on the current and the differential capacity.
    """

    def get_fundamental_variables(self):
        domain, Domain = self.domain_Domain
        phase_name = self.phase_name
        h = pybamm.Variable(
            f"{Domain} electrode {phase_name}hysteresis state",
            domains={
                "primary": f"{domain} electrode",
                "secondary": "current collector",
            },
            bounds=(pybamm.Scalar(-1),pybamm.Scalar(1)),
        )
        return {
            f"{Domain} electrode {phase_name}hysteresis state": h,
        }

    def get_coupled_variables(self, variables):
        domain, Domain = self.domain_Domain
        phase_name = self.phase_name
        phase = self.phase

        if self.reaction == "lithium-ion main":
            T = variables[f"{Domain} electrode temperature [K]"]
            h = variables[f"{Domain} electrode {phase_name}hysteresis state"]

            # For "particle-size distribution" models, take distribution version
            # of c_s_surf that depends on particle size.
            domain_options = getattr(self.options, domain)
            if domain_options["particle size"] == "distribution":
                sto_surf = variables[
                    f"{Domain} {phase_name}particle surface stoichiometry distribution"
                ]
                # If variable was broadcast, take only the orphan
                if isinstance(sto_surf, pybamm.Broadcast) and isinstance(
                    T, pybamm.Broadcast
                ):
                    sto_surf = sto_surf.orphans[0]
                    T = T.orphans[0]
                T = pybamm.PrimaryBroadcast(T, [f"{domain} {phase_name}particle size"])
                h = pybamm.PrimaryBroadcast(h, [f"{domain} {phase_name}particle size"])
            else:
                sto_surf = variables[
                    f"{Domain} {phase_name}particle surface stoichiometry"
                ]
                # If variable was broadcast, take only the orphan
                if isinstance(sto_surf, pybamm.Broadcast) and isinstance(
                    T, pybamm.Broadcast
                ):
                    sto_surf = sto_surf.orphans[0]
                    T = T.orphans[0]

            variables[
                f"{Domain} electrode {phase_name}hysteresis state distribution"
            ] = h

[... 102 lines cut ...]
            Q_cell = variables[f"{Domain} electrode capacity [A.h]"]

        dQdU = variables[
            f"{Domain} electrode {phase_name}differential capacity [A.s.V-1]"
        ]
        dQdU = dQdU.orphans[0]
        K = self.phase_param.hysteresis_decay
        K_x = self.phase_param.hysteresis_switch
        h = variables[f"{Domain} electrode {phase_name}hysteresis state"]

        # dhdt = (
        #     K * (current / (Q_cell * (dQdU**K_x))) * (1 - pybamm.sign(current) * h)
        # )  #! current is backwards for a halfcell
        dhdt = (
            K * (current*L0*acc*a*epss / (Q_cell * (dQdU**K_x))) * (1 - pybamm.sign(current) * h)
        )
        self.rhs[h] = dhdt

    def set_initial_conditions(self, variables):
        domain, Domain = self.domain_Domain
        phase_name = self.phase_name
        h = variables[f"{Domain} electrode {phase_name}hysteresis state"]
        self.initial_conditions[h] = pybamm.Scalar(-1)

```

ejfdickinson, 2024-08-15:
@llh929423719 Could you explain what your changes mean in mathematical terms? I see that you corrected some of the unit peculiarities for the implied units of the `decay rate`. What are the other implications?

@mleot I was working with this submodel recently (thank you for the very helpful addition!). The Plett model and the Wycisk variation defines the decay rate $\gamma$ as (effectively) $\partial \ln h / \partial x_\mathrm{Li}$. I noticed that your $K$ differs as:

$K = \frac{\gamma}{3600} a_\mathrm{vol}V_\mathrm{el}$

where $a_\mathrm{vol}$ is volumetric surface area and $V_\mathrm{el}$ is electrode volume. Hence the decay rate is not unitless but rather in its present form has implied units m^2 (for `K_x == 0` - if `K_x != 0` then normalising `dQdU` to a reference value would be tidier unitwise). I think that revising the expression to use the standard decay rate definition would be better but now raises backwards-compatibility issues.

@llh929423719 I notice that you also changed the initial condition for `h` in `set_initial_conditions()`. @mleot It would be useful to be able to set the initial hysteresis state via the parameter values, like other variable initialisation.

llh929423719, 2024-08-15:
@ejfdickinson The formula in the Wycisk paper is $\frac{\mathrm{d}h(z,t)}{\mathrm{d}t}=\left(\frac{k(z)\cdot I(t)}{Q_{\mathrm{Cell}}}\right)\left(1-\mathrm{sgn}\left(\frac{\mathrm{d}z(t)}{\mathrm{d}t}\right)h(z,t)\right)$ where $Q_{cell}$ is the capacity of the cell and $𝐼(𝑡)$ is the current. As in pybamm, the current $𝐼(𝑡)$ uses the interfacial current density [A.m-2] distribution(which is relatively large), but $Q_{cell}$ uses electrode phase capacity [A.h](which is relatively small), so I multiply the current density by the active material volume fraction $\varepsilon_{s}$ and the total electrode volume

ejfdickinson, 2024-08-15:
Thanks @llh929423719 - I think your approach makes the most sense so that `k` has natural units (unitless) - take the ratio of current to capacity on the same scale, as you would for the full cell.

Ultimately your change just alters the numerical value of the decay rate though, rather than changing the structure of the equations. So you could achieve the same just be scaling the decay rate, right? Do you see the same improvement if you keep the 24.5 release code and scale the decay rate parameter magnitude?

llh929423719, 2024-08-15:
Thanks @ejfdickinson , keeping the 24.5 release code and scale the decay rate parameter magnitude has the same effect.

mleot, 2024-08-16:
Hi @ejfdickinson, thanks for these comments. You make a fair point - as I was writing the code for this, I realized that realistically the current which is impacting the hysteresis should be measured locally at the particle interface, instead of a lumped electrode-level current. I believe I made an oversight by not reflecting and adjusting the units accordingly. I think there could be an argument for having K be a united quantity, although I have not thought about it enough to make any arguments there.

Further, I do like the suggestion of being able to specify the initial hysteresis state condition in the parameter set. 

@llh929423719 I think the modification that you made above is not a bad idea. My only criticism would be that I don't believe the electrode thickness needs to be included in the calculation. I believe the alternate form below would make sure K is unitless.

```
dhdt = (
            K * (current / ((Q_cell / (acc * epss * a)) * (dQdU**K_x))) * (1 - pybamm.sign(current) * h)
        )
        self.rhs[h] = dhdt
```

I suspect the issue you encountered may have been from modifying the initial hysteresis state to a value very close to it's limit (I don't believe the hysteresis state variable should every actually reach 1 or -1). In the original bug, did you modify the initial condition at all?

llh929423719, 2024-08-16:
@mleot thanks for reply, I modify the initial condition just for fit the experiment data (cell is fully charged before discharge), but setting initial condition to zero does not make improvement. I take the electrode thickness into calculation just because the $Q_{cell}$ is calaulated cross the whole electrode, and $I(t) [A.m^{-2}] \cdot L_{0} [m] \cdot A_{cc} [m^2] \cdot a [m^{-1}]$ has the same unit $A$ as in Wycisk's paper Equ[9] 
$\frac{\mathrm{d}h(z,t)}{\mathrm{d}t}=K\cdot\left(\frac{I(t)}{Q_{\mathrm{Cell}}\cdot C_{\mathrm{diff}}(z)}\right)\left(1-\mathrm{sgn}\left(\frac{\mathrm{d}z(t)}{\mathrm{d}t}\right)h(z,t)\right)$. 
But maybe I have misunderstood some of the variables in the pybamm model.

ejfdickinson, 2024-08-16:
@mleot @llh929423719 

To keep electrode thickness out and to keep the focus on particle-scale properties, would it be better to normalise to the volumetric charge capacity of the pure material instead? That is:

$\frac{\partial h}{\partial t} \propto K\cdot\left(\frac{i_\mathrm{surf}a_\mathrm{vol}}{\phi_\mathrm{act}Fc_\mathrm{max}}\right)$

This means that the formulation is considering the particle-scale ratio of current to charge capacity, which would be more appropriate with e.g. blended electrodes in which we are modelling hysteresis for only one component. For example, in a graphite-silicon electrode, if you use the Wycisk model only for the Si and set the hysteresis relaxation rate proportional to the Si interfacial current density (as makes sense), but you normalise the decay rate constant to the whole-electrode capacity, you'd need to keep rescaling the decay rate constant whenever you change the Si active material volume fraction (and hence the ratio of Si capacity to whole-electrode capacity). Otherwise the decay rate constant ceases to be a material-specific property. Of course the decay rate constant arises from particle distribution so it might have some dependence on volume fraction, but it shouldn't do so directly in this way.

rtimms, 2024-08-18:
I’m in favour of updating this formulation. It will be helpful to have reusable and easily interpretable parameters, especially as we look to integrate more hysteresis models.

ejfdickinson, 2024-08-19:
@rtimms Would it be worth opening a discussion for this? I've used the `WyciskOpenCircuitPotential` feature a fair amount in the last couple of months and have a number of suggestions, but better that these are aired before committing too unilaterally.

rtimms, 2024-08-19:
I’m happy to use this issue to discuss the changes. Could you post your suggestions and we can figure out a plan to move forward?

ejfdickinson, 2024-08-19:
@rtimms @mleot @llh929423719 

Suggestions for improvements:

- Solve a general one-state equation:

$\frac{\partial h}{\partial t} = \gamma\cdot\left(\frac{i_\mathrm{surf}a_\mathrm{vol}}{\phi_\mathrm{act}Fc_\mathrm{max}}\right)\cdot \left(1 - \mathrm{sgn}(i_\mathrm{surf})h\right)$

for $i_\mathrm{surf}$ the interfacial phase-specific current density (anodic current density positive).

- Allow a scalar input for $\gamma$, or allow $\gamma$ to be computed using the specific Wycisk formula:

$\gamma = K \cdot \left(\frac{\frac{\partial q_\mathrm{vol}}{\partial U}}{\left.\frac{\partial q_\mathrm{vol}}{\partial U}\right|_\mathrm{ref}}\right)^{-n}$

where the evaluation at a reference point (implicitly at $x_\mathrm{Li}=x_\mathrm{Li,ref}$ which in turn defines $K$ as the decay rate at the reference stoichiometry) is to ensure that we take an arbitrary power only of a unitless quantity.

- Allow general (particle-wise) initialisation of the hysteresis state, not necessarily = 0.

mleot, 2024-08-19:
@ejfdickinson Thanks for the suggestions. 

Regarding your first suggestion: I think is a neat implementation of this portion of the equation. 

Regarding the second suggestion: Is this specific equation referenced in the Wycisk paper? This does seem like a better way to implement the differential capacity with the power term. How do you suggest the reference point be implemented? An additional parameter for a specific stoichiometry from which the reference value is extracted?

I also like the idea of being able to specify a custom function, but the implementation for that seems potentially challenging. It might be nice in an alternate implementation to have a $$\gamma$$ which accounts for rate invariance (mentioned in the Baker 2017 paper)[link](https://pubs.aip.org/aip/jap/article-abstract/122/16/165102/149622/An-approach-to-characterize-and-clarify-hysteresis?redirectedFrom=fulltext).

The last suggestion makes sense and I have no arguments against it.

ejfdickinson, 2024-08-20:
@mleot I don't think this specific equation is anywhere in Wycisk but it's just a factorisation of a scalar into the product of two parts with the effect of (a) tidying up the units and (b) correspondingly making more explicit and comprehensible the physical meaning of the parameters. Wycisk et al. would be far from the first authors to fail to introduce the tidiest mathematical expression of their own good idea, particularly when a general-purpose modelling approach is envisaged.

Setting the specific stoichiometry at which the reference value is extracted is sensible to me. That just changes the definition of `K` from "decay rate" to "decay rate at xLi = ???", where the Wycisk theory then uses (dq/dU)^(-n) as the scaling to get decay rate at other xLi.

ejfdickinson, 2024-08-20:
I don't completely know enough about PyBaMM's architecture to know the difficulties of arbitrary funtional input for $\gamma$ once this parameter is exposed in the form I wrote it.

At least a scalar input option would be very useful as an 'opt-out' from the specific Wycisk relation. I have some models at the moment where I achieve scalar input by setting `n == 0`, but the existing implementation still always tries to evaluate $dq/dU$ so some shenanigans are needed to work around cases where `_function_diff` is undefined for the open-circuit potential expression (e.g. interpolants).

mleot, 2024-08-21:
Referring to @rtimms for advice on difficulty of implementation

rtimms, 2024-08-22:
If you make something a `FunctionParameter`, it can be a function or a parameter. The only thing that is fixed is the arguments of the function. What variables could $\gamma$ reasonably depend on?

ejfdickinson, 2024-08-23:
@rtimms @mleot I guess in the first instance we would expect this to be parameterised depending on lithiation extent (cs / csmax) and temperature, like diffusivity is.

Anything more would be in the realm of future custom submodels depending on academic developments in understanding of the physical phenomena that dictate the magnitude of $\gamma$. In reality I think it's an inter-particle phenomenon for LFP and probably fairly mechanically controlled for Si, so there's loads of extra physics if you want to get beyond semi-empiricism. That's definitely for the future!

rtimms, 2024-08-23:
Sounds good! Would either of you like to take this on? IMO we should just do it as a breaking change and make sure we document how to update parameters that users have for the current implementation to match the new one.

ejfdickinson, 2024-08-23:
I would recommend that @mleot continues actual development of this class (if he is willing!), but I would be happy to assist with code review and testing.

rtimms, 2024-08-23:
If helpful, I’m happy to help with setting up the new class structure. Then I’ll leave the details to @mleot (again, if you’re willing!).

mleot, 2024-08-23:
I would be happy to help! @rtimms If you want to set up the new class structure that would be very helpful!

mleot, 2024-12-27:
I had some trouble implementing the custom function parameters mentioned in this issue. Implementing the initial state was fairly easy to implement, however. I will propose that the initial hysteresis state be a separate PR from the custom function parameter implementation. I will open up a different issue for just implementing the initial hysteresis state. 

See #4714 for the new issue.