pybamm-team/PyBaMM #4338: Extend `pybamm.ParameterValues.create_from_bpx()` to automate initial concentrations for blended electrodes

ejfdickinson, 2024-08-13:
### Description

Calling `pybamm.ParameterValues.create_from_bpx()` with a BPX file containing a blended electrode returns a warning as follows:

`Initial concentrations cannot be set using stoichiometry limits for blend electrodes. Please set the initial concentrations manually.`

The method could automate this calculation by solving the equations specified in the BPX standard.

rtimms, 2024-08-13:
Tagging https://github.com/pybamm-team/PyBaMM/issues/2682 as related. Currently the electrode state of health models in pybamm don't support blended electrodes.

ejfdickinson, 2024-08-14:
@rtimms Off-topic I know, but for this reason I find it odd that `calc_esoh=True` is the default in `pybamm.Simulation.solve()`.

rtimms, 2024-08-15:
Yes, this is simply because `calc_esoh` predates blended electrodes. I’m open to changing this.

valentinsulzer, 2024-08-16:
The proper fix, which we are planning to do, is to do "just in time" calculations of esoh values only when they are requested