pybamm-team/PyBaMM #4343: [Bug]: Use of lead-acid battery models

simdieudoboinzemwende, 2024-08-14:
### PyBaMM Version

24.5

### Python Version

3.12.3

### Describe the bug

First of all, I mainly work on lithium batteries, but for some reason, I want to conduct an experiment to simulate some aging mechanisms of a lead-acid battery this time. The approach I usually take here is a long experiment with charge and discharge cycles. However, this time I am encountering errors. Apparently, with lead-acid models, we cannot use cut-off voltages as conditions to delimit the cycles.
I am really stuck at this point because if I use time as the end-of-cycle condition, it fails over time due to overlaps.

### Steps to Reproduce

model = pybamm.lead_acid.Full(options={'thermal':'lumped'})
params = model.default_parameter_values
solver = pybamm.CasadiSolver(mode="fast", extra_options_setup={"max_num_steps": 1000})
params['Ambient temperature [K]']=298.15
sim = pybamm.Simulation(model, parameter_values=params, experiment=experiment_long, solver=solver)
sol = sim.solve()

Here is the first approach:
experiment_long = pybamm.Experiment(
    [
        (
            "Discharge at C/10 until 1.8V",
            "Rest for 10 minutes",
            "Charge at 1C until 2.4V",
            "Hold at 2.4V until C/50",
        )
    ]
    * 500
)

Here is the second approach:
experiment_long = pybamm.Experiment(
    [
        (
            "Discharge at C/10 for 10 hours",
            "Rest for 10 minutes",
            "Charge at 1C for 1 hours",
        )
    ]
    * 300
)

### Relevant log output

```shell
erreur for the first approach:
2024-08-14 16:47:52.394 - [NOTICE] logger.func(7): Cycle 1/500 (17.300 us elapsed) --------------------
2024-08-14 16:47:52.395 - [NOTICE] logger.func(7): Cycle 1/500, step 1/4: Discharge at C/10 until 1.8V
At t = 48137.4, repeated recoverable residual errors.
2024-08-14 16:48:04.871 - [ERROR] callbacks.on_experiment_error(233): Simulation error: Error in Function::call for 'F' [IdasInterface] at .../casadi/core/function.cpp:1432:
Error in Function::call for 'F' [IdasInterface] at .../casadi/core/function.cpp:361:
.../casadi/interfaces/sundials/idas_interface.cpp:596: IDASolve returned "IDA_REP_RES_ERR". Consult IDAS documentation.
SolverError: Error in Function::call for 'F' [IdasInterface] at .../casadi/core/function.cpp:1432:
Error in Function::call for 'F' [IdasInterface] at .../casadi/core/function.cpp:361:
.../casadi/interfaces/sundials/idas_interface.cpp:596: IDASolve returned "IDA_REP_RES_ERR". Consult IDAS documentation.

erreur for the second approach:
Experiment is infeasible: 'event: Max positive electrode porosity cut-off' was triggered during 'Charge at 1C for 1 hours'. The returned solution only contains up to step 3 of cycle 15.
```

valentinsulzer, 2024-08-16:
The casadi solver ignores events if you use "fast" mode.
This works
```python
import pybamm

model = pybamm.lead_acid.Full(options={"thermal": "lumped"})
params = model.default_parameter_values
solver = pybamm.CasadiSolver(extra_options_setup={"max_num_steps": 1000})
params["Ambient temperature [K]"] = 298.15

experiment_long = pybamm.Experiment([("Discharge at 1C until 10.5V")])

sim = pybamm.Simulation(
    model, parameter_values=params, experiment=experiment_long, solver=solver
)
sol = sim.solve()
sol.plot(["Voltage [V]", "Battery voltage [V]"])
```

simdieudoboinzemwende, 2024-08-18:
Thank you so much for the assistance

closed 2024-08-19 by valentinsulzer