pybamm-team/PyBaMM #4355: Variable problem of lithium_ionModel

Santiagopeacely, 2024-08-19:
pybamm.lithium_ion.DFN()，is dividing the two variables related to the heat generation of the positive and negative current collectors by the conductivity multiplied by the conductivity?
[image]

brosaplanella, 2024-08-27:
Can you provide a minimum working example about this bug?

MarcinK00, 2024-12-03:
The calculations are correct. For example, for a 10 um thick copper collector at a current density of 3 mA/cm2 (one side coat), the ohmic heating is 1.55E-5 W/m3. At the same time, the value completely disagrees with reality. It should be several orders of magnitude higher. This is due to a simple reason. The calculated ohmic heating is for conduction through the (collector) plane; in practice, the heating is related to conduction in the (collector) plane. Getting the correct value would require solving an additional PDE in 2D. I believe that as it stands, ohmic heating for current collectors is a bit misleading, but I guess it is required.