pybamm-team/PyBaMM #4365: [Bug]: Noise in low-current results using O'Kane 2022 parameter set due to graphite OCP interpolation from csv file

tomjholland, 2024-08-21:
### PyBaMM Version

24.5

### Python Version

3.11.7

### Describe the bug

Differentiating low c-rate constant-current results reveals noise in calculated voltage when using `"OKane2022"` parameter set, but not when using `"Chen2020"` parameter set:
[image]

This is due to the csv interpolation used for the graphite OCP in the `"OKane2022"` parameter set. The suggested fix would be to replace this with the functional form defined in `"Chen2020"`, as the OCP curves for these parameter sets should be identical anyway.

### Steps to Reproduce

The following code uses the Chen2020 parameters, OKane2022 parameters and finally the OKane2022 parameters with the graphite OCP parameter replaced with the value in the Chen2020 parameter set:

```python
import pybamm
import numpy as np
import matplotlib.pyplot as plt

parameter_sets = ["OKane2022", "Chen2020", "OKane2022 OCP Function"]
C_rate = 0.01
plt.figure()
for parameter_set in parameter_sets:
    if parameter_set == "OKane2022 OCP Function":
        params = pybamm.ParameterValues("OKane2022")
        params["Negative electrode OCP [V]"] = pybamm.ParameterValues("Chen2020")['Negative electrode OCP [V]']
    else:
        params = pybamm.ParameterValues(parameter_set)  
    params.set_initial_stoichiometries(0)

    model = pybamm.lithium_ion.DFN()
    solver = pybamm.CasadiSolver()
    experiment = pybamm.Experiment([f"Charge at {C_rate}C until 4.2 V"])
    sim = pybamm.Simulation(
        model,
        solver=solver,
        parameter_values=params,
        experiment=experiment,
    )

    sim.solve()

    dVdQ = np.gradient(sim.solution["Terminal voltage [V]"].entries, sim.solution["Discharge capacity [A.h]"].entries)

    plt.plot(sim.solution["Discharge capacity [A.h]"].entries, dVdQ, label=parameter_set)

plt.xlabel("Capacity [Ah]")
plt.ylabel("dV/dQ [V/Ah]")
plt.legend()
plt.show()

rtimms, 2024-08-21:
can you try using a linear interpolant rather than a cubic one? something like 

```python3
parameter_values = pybamm.ParameterValues("Okane2022")

path = # path to where the csv is stored
graphite_LGM50_ocp_Chen2020_data = pybamm.parameters.process_1D_data(
    "graphite_LGM50_ocp_Chen2020.csv", path=path
)

def graphite_LGM50_ocp_Chen2020(sto):
    name, (x, y) = graphite_LGM50_ocp_Chen2020_data
    return pybamm.Interpolant(x, y, sto, name=name, interpolator="linear")

parameter_values["Negative electrode OCP [V]"] = graphite_LGM50_ocp_Chen2020
```

I think all the data interpolant OCPs should be linear, but it looks like they aren't...

kratman, 2024-08-22:
> can you try using a linear interpolant rather than a cubic one?

I tried this yesterday, still has the noise with linear interpolation. I also checked and several other parameter sets use cubic interpolation:
[image]

I put a couple of my experiments below.

### OKane2022 with linear

[image]

### OKane2022 with linear and graphite_ocp_Enertech_Ai2020

[image]

tomjholland, 2024-08-23:
> can you try using a linear interpolant rather than a cubic one? something like
> 
> ```python
> parameter_values = pybamm.ParameterValues("Okane2022")
> 
> path = # path to where the csv is stored
> graphite_LGM50_ocp_Chen2020_data = pybamm.parameters.process_1D_data(
>     "graphite_LGM50_ocp_Chen2020.csv", path=path
> )
> 
> 
> def graphite_LGM50_ocp_Chen2020(sto):
>     name, (x, y) = graphite_LGM50_ocp_Chen2020_data
>     return pybamm.Interpolant(x, y, sto, name=name, interpolator="linear")
> 
> parameter_values["Negative electrode OCP [V]"] = graphite_LGM50_ocp_Chen2020
> ```
> 
> I think all the data interpolant OCPs should be linear, but it looks like they aren't...

Yes so the results may be dependent on the interpolator (thanks @kratman for looking into this), my question I think is more fundamental. To my understanding:
- The `"OKane2022"` parameter set is based on the `"Chen2020"` parameters for the LG M50 cell
- The `"Chen2020"` parameter set uses fitted functions for _both_ positive and negative OCPs, as defined in their paper
- `"OKane2022"` also uses a _fitted function_ for the positive OCP
-  `"OKane2022"` uses _interpolation_ for the negative OCP

I raised this issue to suggest that the `"OKane2022"` is brought in line with `"Chen2020"` parameter set to use the fitted function form for consistency, although I realise that this is not a general solution for the other parameter sets that just use interpolation.

kratman, 2024-08-23:
Yeah I used the Ai2020 data to show that something was specifically an issue with the data set used by OKane2022. It could be as simple as not having enough resolution to make the function smooth

rtimms, 2024-11-22:
Since this was the OCP used in that paper I think we should leave it as-is so that the parameter set stays true to the reference.

kratman, 2024-11-27:
Since we are leaving this with the original OCP, I am going to close this ticket

closed 2024-11-27 by kratman