pybamm-team/PyBaMM #4407: Add CVODE solver

MarcBerliner, 2024-09-03:
### Description

From https://github.com/pybamm-team/PyBaMM/issues/3910#issuecomment-2312804655

To improve the speed of our ODE models, I'd like to add `CVODE` from SUNDIALS to our suite of C solvers in addition to `IDA`. I have a few questions about the implementation, and I'd like to hear your thoughts @jsbrittain, @martinjrobins, @pipliggins, and others.

- In C++, we can make a base solver class and derived classes for `IDA` and `CVODE`. The solver code structure for both will be very similar, so this will help with code reuse. However, this will cause a headache for some of the existing PRs.
- The `CasadiSolver` automatically determines if the system is an ODE or a DAE and passes it to the appropriate solver. I think this approach also makes sense for our C solvers. If we do take this approach, then...
- One minor issue is that the `IDAKLUSolver` name will no longer be accurate. Since we plan on streamlining the `IDAKLU` installation process and changing the default away from `CasadiSolver`, maybe we can also rename `IDAKLUSolver` to `SundialsSolver` or something. This might make the "new and improved default solver" announcement splashier (cc @valentinsulzer @rtimms).

And just FYI, I don't plan on starting this work for at least a couple of weeks.

martinjrobins, 2024-09-13:
I prefer the new name you suggested `SundialsSolver` (instead of `IDAKLUSolver`), makes sense. Have you started on this one yet and is there a branch I can have a look at ? I'd like to implement openmp parallisation for the `SundialsSolver` sometime soon (i.e. part of https://github.com/pybamm-team/PyBaMM/pull/4260) but can hold off if you have already started

MarcBerliner, 2024-09-13:
> I prefer the new name you suggested `SundialsSolver` (instead of `IDAKLUSolver`), makes sense. Have you started on this one yet and is there a branch I can have a look at ? I'd like to implement [...]

I haven't started working on this yet.

agriyakhetarpal, 2025-01-13:
I came across this project some time ago, which might be helpful: https://github.com/bjodah/pycvodes

agriyakhetarpal, 2025-01-13:
Also, I guess it makes sense to transfer the issue to `pybammsolvers` now, doing so