pybamm-team/PyBaMM #4414: IDAKLU-JAX not working with BPX model

MarkV-ADI, 2024-09-04:
Hi all, I recently been playing around with the IDAKLU-JAX Solver and very impressed with it's speed! I am trying to target a BPX model cell. However when i do, i get this error: KeyError: 'Current function [A]'

```
import pybamm
import numpy as np
import time

# Define inputs
inputs = {
    "Current function [A]": 0.222,
    "Separator porosity": 0.3,
}

# Set-up the model
options = {"cell geometry": "arbitrary", "thermal": "lumped"}
model = pybamm.lithium_ion.DFN(options=options)
geometry = model.default_geometry
# param = model.default_parameter_values
param = pybamm.ParameterValues.create_from_bpx(r'lfp_18650_cell_BPX.json')
param.update(inputs)
param.process_geometry(geometry)
param.process_model(model)
var = pybamm.standard_spatial_vars
var_pts = {var.x_n: 20, var.x_s: 20, var.x_p: 20, var.r_n: 10, var.r_p: 10}
mesh = pybamm.Mesh(geometry, model.default_submesh_types, var_pts)
disc = pybamm.Discretisation(mesh, model.default_spatial_methods)
disc.process_model(model)

# Use a short time-vector for this example
t_eval = np.linspace(0, 360, 10)

# Include the variable of interest in output_variables
output_variables = [
    "Voltage [V]",
    "Current [A]",
    "Time [min]",
    "X-averaged cell temperature [C]" # want to retrieve this and print the output
]

# Create the IDAKLU Solver object
idaklu_solver = pybamm.IDAKLUSolver(rtol=1e-6, atol=1e-6, output_variables=output_variables)

start_time = time.time()

# Perform the simulation
sim = idaklu_solver.solve(
    model,
    t_eval,
    inputs=inputs,
    calculate_sensitivities=True,
)

end_time = time.time()
print(f"Simulation took {end_time - start_time:.2f} seconds.")

# Variable name to extract
var_name = "X-averaged cell temperature [C]"

# Check if the variable is present in the simulation results
if var_name not in sim._variables:
    print(f"Variable '{var_name}' not found in the simulation results.")
else:
    # Extract the variable from the simulation result
    try:
        # Evaluate the variable using the simulation result
        temp_data = sim[var_name](t_eval)
        print("X-averaged cell temperature [C]:")
        print(temp_data)
    except Exception as e:
        print(f"An error occurred while evaluating the variable: {e}")
```

The BPX file used is found from the about:energy repo for BPX: https://github.com/About-Energy-OpenSource/About-Energy-BPX-Parameterisation/blob/main/LFP/lfp_18650_cell_BPX.json

I thought that i could add in the current function parameter similar to how it was done in chen2020 but after modifying the bpx parameterization, cell to have a current function
```
"Parameterisation": {
            "Cell": {
                  "Ambient temperature [K]": 298.15,
                  "Initial temperature [K]": 298.15,
                  "Reference temperature [K]": 298.15,
                  "Lower voltage cut-off [V]": 2.0,
                  "Upper voltage cut-off [V]": 3.65,
                  "Nominal cell capacity [A.h]": 2,
                  "Specific heat capacity [J.K-1.kg-1]": 999,
                  "Thermal conductivity [W.m-1.K-1]": 1.89,
                  "Density [kg.m-3]": 1940,
                  "Electrode area [m2]": 0.08959998,
                  "Number of electrode pairs connected in parallel to make a cell": 1,
                  "External surface area [m2]": 0.00431,
                  "Volume [m3]": 1.7e-05,
                  "Current function [A]": 5
            },
```
my response to the code is
```
  extra fields not permitted (type=value_error.extra)
Parameterisation -> Separator
  extra fields not permitted (type=value_error.extra)
Parameterisation -> Cell -> Current function [A]
  extra fields not permitted (type=value_error.extra).....
```
so on and so on for each parameter...

Is there a way i can use idaklu solver with the bpx model?
should my input dict change to not use current?, am i meant to modify the bpx file somehow to target current or is there a general recomend way for me to solve with IDAKULU for JAX.

Thanks all in advance :)

MarkV-ADI, 2024-09-05:
Hi - I think i found out how to resolve what i was trying to simulate by finding an old discussion here: https://github.com/pybamm-team/PyBaMM/discussions/4296

```
import pybamm
import os
import pandas as pd
from pybamm.expression_tree.functions import exp
import numpy as np

pybamm.set_logging_level("INFO")

experiment = pybamm.Experiment(
    [
        (
            "Discharge at C/5 for 10 hours or until 2.5 V",
        )
    ]
)

options = {"thermal": "lumped"}
model = pybamm.lithium_ion.DFN(options=options)
parameter_values = pybamm.ParameterValues.create_from_bpx(r'lfp_18650_cell_BPX.json')

t_eval = np.linspace(0, 3600, 100)

solver = pybamm.IDAKLUSolver()
sim = pybamm.Simulation(model,parameter_values=parameter_values, solver=solver, experiment=experiment)
sim.solve()
output_variables = ["Voltage [V]", "Current [A]", 'X-averaged cell temperature [C]']
sim.plot(output_variables=output_variables)
```

### **Result**
```
2024-09-05 14:21:13.416 - [INFO] base_model._build_model(777): Start building Doyle-Fuller-Newman model
2024-09-05 14:21:13.508 - [INFO] base_battery_model.build_model(1042): Finish building Doyle-Fuller-Newman model
2024-09-05 14:21:13.584 - [INFO] callbacks.on_experiment_start(162): Start running experiment
2024-09-05 14:21:13.587 - [INFO] parameter_values.process_model(440): Start setting parameters for Doyle-Fuller-Newman model
2024-09-05 14:21:13.723 - [INFO] parameter_values.process_model(509): Finish setting parameters for Doyle-Fuller-Newman model
2024-09-05 14:21:13.727 - [INFO] discretisation.process_model(149): Start discretising Doyle-Fuller-Newman model
2024-09-05 14:21:14.440 - [INFO] discretisation.process_model(251): Finish discretising Doyle-Fuller-Newman model
2024-09-05 14:21:14.443 - [NOTICE] callbacks.on_cycle_start(170): Cycle 1/1 (107.200 us elapsed) --------------------
2024-09-05 14:21:14.445 - [NOTICE] callbacks.on_step_start(178): Cycle 1/1, step 1/1: Discharge at C/5 for 10 hours or until 2.5 V
2024-09-05 14:21:14.454 - [INFO] base_solver.set_up(117): Start solver set-up
2024-09-05 14:21:14.953 - [INFO] base_solver.set_up(286): Finish solver set-up
2024-09-05 14:21:15.999 - [INFO] parameter_values.process_model(440): Start setting parameters for ElectrodeSOH model
2024-09-05 14:21:16.015 - [INFO] parameter_values.process_model(509): Finish setting parameters for ElectrodeSOH model
2024-09-05 14:21:16.016 - [INFO] discretisation.process_model(149): Start discretising ElectrodeSOH model
2024-09-05 14:21:16.034 - [INFO] discretisation.process_model(251): Finish discretising ElectrodeSOH model
2024-09-05 14:21:16.038 - [INFO] base_solver.solve(744): Start solving ElectrodeSOH model with Algebraic solver (lm)
2024-09-05 14:21:16.042 - [INFO] base_solver.set_up(117): Start solver set-up
2024-09-05 14:21:16.060 - [INFO] base_solver.set_up(286): Finish solver set-up
2024-09-05 14:21:16.066 - [INFO] base_solver.solve(974): Finish solving ElectrodeSOH model (the solver successfully reached the end of the integration interval)
2024-09-05 14:21:16.068 - [INFO] base_solver.solve(975): Set-up time: 19.615 ms, Solve time: 5.251 ms (of which integration time: 4.703 ms), Total time: 24.866 ms
2024-09-05 14:21:16.124 - [NOTICE] callbacks.on_experiment_end(229): Finish experiment simulation, took 1.681 s
```

I just have one question if anyone gets around to this...
Why is it when setting the IDAKLU solver here solver with 'Algebraic solver (lm)'
but if i solved for ```t_eval = np.linspace(0, 3600, 100)```

my results is:
```
2024-09-05 14:24:43.731 - [INFO] base_model._build_model(777): Start building Doyle-Fuller-Newman model
2024-09-05 14:24:43.826 - [INFO] base_battery_model.build_model(1042): Finish building Doyle-Fuller-Newman model
2024-09-05 14:24:43.889 - [INFO] parameter_values.process_model(440): Start setting parameters for Doyle-Fuller-Newman model
2024-09-05 14:24:43.985 - [INFO] parameter_values.process_model(509): Finish setting parameters for Doyle-Fuller-Newman model
2024-09-05 14:24:43.987 - [INFO] discretisation.process_model(149): Start discretising Doyle-Fuller-Newman model
2024-09-05 14:24:44.786 - [INFO] discretisation.process_model(251): Finish discretising Doyle-Fuller-Newman model
2024-09-05 14:24:44.787 - [INFO] base_solver.solve(744): Start solving Doyle-Fuller-Newman model with IDA KLU solver
2024-09-05 14:24:44.792 - [INFO] base_solver.set_up(117): Start solver set-up
2024-09-05 14:24:45.224 - [INFO] base_solver.set_up(286): Finish solver set-up
2024-09-05 14:24:46.016 - [INFO] base_solver.solve(974): Finish solving Doyle-Fuller-Newman model (the solver successfully reached the end of the integration interval)
2024-09-05 14:24:46.017 - [INFO] base_solver.solve(975): Set-up time: 615.222 ms, Solve time: 607.619 ms (of which integration time: 596.279 ms), Total time: 1.223 s
```

Regardless both time eval and experiment with idaklu are significantly faster than casadi but still would like to know the difference between using idaklu to solve for time and for an experiment :)

ejfdickinson, 2024-09-12:
@MarkV-ADI In relation to the BPX issue, you can't write in new fields in this part of the BPX file or it will be rejected by the schema. The schema errors themselves are not helpful (https://github.com/FaradayInstitution/BPX/issues/52).

BPX permits you to add any additional fields for the BPX file under `"Parameterisation" -> "User-defined"`. PyBaMM imports these directly into the `pybamm.ParameterValues` object, by default.

Does this error persist if you apply your inputs to the parameter values (as you are doing), assemble a `pybamm.Simulation` directly and let PyBaMM do the default things, rather than processing the model in place? This might help debug if there is still a PyBaMM issue here. I've put some approximate code below.

```
# Set-up the model
options = {"cell geometry": "arbitrary", "thermal": "lumped"}
model = pybamm.lithium_ion.DFN(options=options)
geometry = model.default_geometry
# param = model.default_parameter_values
param = pybamm.ParameterValues.create_from_bpx(r'lfp_18650_cell_BPX.json')
param.update(inputs)
var = pybamm.standard_spatial_vars
var_pts = {var.x_n: 20, var.x_s: 20, var.x_p: 20, var.r_n: 10, var.r_p: 10}

# Use a short time-vector for this example
t_eval = np.linspace(0, 360, 10)

# Include the variable of interest in output_variables
output_variables = [
    "Voltage [V]",
    "Current [A]",
    "Time [min]",
    "X-averaged cell temperature [C]" # want to retrieve this and print the output
]

# Create the IDAKLU Solver object
idaklu_solver = pybamm.IDAKLUSolver(rtol=1e-6, atol=1e-6, output_variables=output_variables)

sim = pybamm.Simulation(
    model,
    parameter_values=param,
    geometry=geometry,
    var_pts=var_pts,
    solver=idaklu_solver,
    output_variables=output_variables,
)
sol = sim.solve(
    t_eval=t_eval,
    calculate_sensitivities=True,
)
```

pipliggins, 2024-09-13:
@MarkV-ADI @ejfdickinson's code should work if you don't use the `output_variables` option in the IDAKLU solver. Using the output_variables option reveals a small bug I've noted in #4439.

kratman, 2024-10-22:
@MarkV-ADI Did this resolve your question?

closed 2024-11-21 by kratman