pybamm-team/PyBaMM #4419: Post-solve Hermite interpolation in Sundials solvers

MarcBerliner, 2024-09-05:
### Description

This will give higher accuracy interpolation for observables after solving the system of equations.

martinjrobins, 2024-09-13:
@MarcBerliner can you add more details on how you see this working? At the moment the `Solution` object and all its post-processing is very separate from the solvers, but it sounds like this would couple the two? What if the idaklu solver or sundials is not installed? Is sundials the best thing to use here or we could store the derivative info in `Solution` and use some other python library that has hermite interpolation (https://numpy.org/doc/stable/reference/generated/numpy.polynomial.hermite.hermfit.html).

MarcBerliner, 2024-09-13:
> @MarcBerliner can you add more details on how you see this working? At the moment the `Solution` object and all its post-processing is very separate from the solvers, but it sounds like this would [...]

Within the ODE/DAE solve, this will add a `hermite_interpolation` option that if true, saves the `yp` and `ySp` terms. We can save these with the `Solution` objects as we do with the `all_ys` and whatnot. The flow of information from solution -> processed variables will be the same. Writing a cubic hermite interpolation script is not too difficult, so I plan on writing this in C++. 

At the moment, observing model or output variables is really slow because we're running a python for-loop over each time series value. I plan on making a "batch observable" function in C++ which will speed up all the post-solve interpolation, but for linear and Hermite interpolation.

closed 2024-10-02 by MarcBerliner