pybamm-team/PyBaMM #4429: [Bug]: pybamm.ScipySolver fails to run experiment

martinjrobins, 2024-09-09:
### PyBaMM Version

develop

### Python Version

3.10

### Describe the bug

the `pybamm.ScipySolver` aborts early on the experiment given below, whereas the casadi and idaklu solver solve it correctly

### Steps to Reproduce

```python
import pybamm
import numpy as np
import matplotlib.pyplot as plt

experiment_2step = pybamm.Experiment(
    [
        (
            "Discharge at C/20 for 1 hour",
            "Charge at 1 A until 4.1 V",
            "Hold at 4.1 V until C/2",
            "Discharge at 2 W for 30 min",
            "Discharge at 2 W for 30 min",  # repeat to cover this case (changes initialisation)
        ),
    ]
    * 2,
)

solutions = []
for solver in [
    pybamm.CasadiSolver(),
    pybamm.IDAKLUSolver(),
    pybamm.ScipySolver(),
]:
      model = pybamm.lithium_ion.SPM()
      param = model.default_parameter_values
      input_param_name = "Negative electrode active material volume fraction"
      input_param_value = param[input_param_name]
      param.update({input_param_name: "[input]"})
      sim = pybamm.Simulation(
          model,
          experiment=experiment_2step,
          solver=solver,
          parameter_values=param,
      )
      solution = sim.solve(
          inputs={input_param_name: input_param_value},
      )
      solutions.append(solution)

for i, s in enumerate(solutions):
    plt.plot(s.t, s["Voltage [V]"](s.t) + float(i), label=f"Solver {i}")
plt.savefig("test.png")

for i in range(1, len(solutions)):
    np.testing.assert_allclose(
        solutions[0]["Voltage [V]"].data[:-1],
        solutions[i]["Voltage [V]"](solutions[0].t[:-1]),
        rtol=5e-2,
        equal_nan=True,
    )
```

[image]

rtimms, 2024-09-09:
This is because voltage control turns the model from an ODE to a DAE. Surprised we don't get a better error message here?

martinjrobins, 2024-09-09:
ahh yes, and the scipy solver doesn't support DAEs. Ok, so the main thing here is raising an error (and updating the relevent test in   #4415)

closed 2024-09-13 by martinjrobins