pybamm-team/PyBaMM #4438: Define `__iter__` and/or `__contains__` for `pybamm.ParameterValues`

ejfdickinson, 2024-09-13:
### Description

The `pybamm.ParameterValues` is dict-like, and through its `__getitem__` and `__setitem__` methods allows dict-like key access. However, it does not support the `in` keyword, which for a dict will treat the keys as an iterable. This means that natural code like:

```python
if "Negative particle radius [m]" in parameter_values:
    # do something
```

will not work (generally raises a `KeyError`). It is currently necessary to instead write explicitly:

```python
if "Negative particle radius [m]" in parameter_values.keys():
    # do something
```

Defining the missing method `__iter__` (and `__contains__` if preferred) would allow `in` to work in logical test and for-loop contexts.

### Motivation

Code is cleaner if `pybamm.ParameterValues` looks like a souped-up dict always, not just sometimes. Key-value accessors are expected to iterate over their keys and to allow logical test on the presence/absence of keys. PyBaMM-containing code looks 'asymmetric' compared to code working with actual `dict` objects due to the requirement to call into `pybamm.ParameterValues.keys()` specifically.

Was there any historic reason for the decision to exclude these methods?

rtimms, 2024-09-13:
AFAIK these weren't intentionally excluded.

martinjrobins, 2024-09-13:
I just ran into this problem today, please do implement these, it would be very helpful! :)

kratman, 2024-09-16:
Yeah I have wanted this as well, but never go around to adding anything

AbhishekChaudharii, 2024-09-24:
Can I take this one?

agriyakhetarpal, 2024-09-24:
Sure, @AbhishekChaudharii, please feel free to try it!

PR #4465 (2024-09-25): Added __iter__ and __contains__ method to pybamm.ParameterValues
# Description

Defined `__iter__ `and `__contains__ `method to `pybamm.ParameterValues` which would allow `in` to work in logical test and for-loop contexts. 

Fixes #4438 

## Type of change

- [x] New feature (non-breaking change which adds functionality)

# Key checklist:

- [x] No style issues: `$ pre-commit run` (or `$ nox -s pre-commit`) (see [CONTRIBUTING.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CONTRIBUTING.md#installing-and-using-pre-commit) for how to set this up to run automatically when committing locall [... cut]
closed 2024-09-25 by arjxn-py