pybamm-team/PyBaMM #4459: [Bug]: KeyError: "'Capacity [A.h]' not found. Best matches are ['Measured capacity [A.h]', 'Throughput capacity [A.h]', 'Change in throughput capacity [A.h]']"

Yull-1, 2024-09-23:
### PyBaMM Version

24.1

### Python Version

3.8.10

### Describe the bug

Subject: KeyError: 'Capacity [A.h]' not found in PyBaMM

Hi PyBaMM team,

I encountered the following error while running my simulation:
`KeyError: "'Capacity [A.h]' not found. Best matches are ['Measured capacity [A.h]', 'Throughput capacity [A.h]', 'Change in throughput capacity [A.h]']`"
Could you advise on how to resolve this issue? Which variable should I use for capacity-related calculations instead of 'Capacity [A.h]'?

While I'm asking, I would also like to know how the values for the following parameters are typically obtained:
```
Initial concentration in positive electrode [mol.m-3]
Initial concentration in negative electrode [mol.m-3]
```
Are these values measured experimentally, and if so, what method is used? Or are they calculated based on other parameters?

Thank you for your assistance!

### Steps to Reproduce

I encountered a KeyError: 'Capacity [A.h]' when running my simulation in PyBaMM. Below is the simplified code that reproduces the issue:

```
import pybamm
import matplotlib.pyplot as plt
import os

model = pybamm.lithium_ion.DFN(
    {
        
        "SEI": "solvent-diffusion limited",
      #  "SEI porosity change": "true",
       # "lithium plating": "partially reversible",
       # "lithium plating porosity change": "false",  # alias for "SEI porosity change"
        #"particle mechanics": ("swelling and cracking",'swelling only'),
       # "SEI on cracks": "true",
       # "loss of active material": "stress-driven",
       # "calculate discharge energy": "false",
    }
)

parameter_values = pybamm.ParameterValues('OKane2022')
parameter_values.update(
	{ 
	#'SEI resistivity [Ohm.m]': 160000.0,
	#'SEI growth activation energy [J.mol-1]': 43000.0,#
	#'Electrode height [m]': 0.0775,
	#'Electrode width [m]': 2.97,
	#'Positive electrode thickness [m]': 6.085e-05,
	#'Maximum concentration in negative electrode [mol.m-3]': 69718.8129,
	#'Maximum concentration in positive electrode [mol.m-3]': 33319,
	#'Separator thickness [m]': 1.5e-05,
    #'Nominal cell capacity [A.h]': 9.73,
	#'Current function [A]': 9.73,
	#'Initial concentration in positive electrode [mol.m-3]': 11100 ,
	#'Initial concentration in negative electrode [mol.m-3]': 57726.13,
	#'Positive electrode conductivity [S.m-1]': 3.163,
	#'Positive electrode porosity': 0.286,
	#'Separator porosity': 0.38,
	#'Positive electrode active material volume fraction': 0.7137,
	#'Negative particle radius [m]': 4.369e-6,
	 }
	)
# Load negative custom OCP data
path, _ = os.path.split(os.path.abspath(__file__))
my_ocp_data = pybamm.parameters.process_1D_data("my_ocp_function.csv", path=path)
# Define custom OCP function
def my_ocp_function(sto):
    name, (x, y) = my_ocp_data
    #print(name,(x,y))
    return pybamm.Interpolant(x, y, sto, name=name, interpolator="cubic")
#parameter_values['Negative electrode OCP [V]'] = my_ocp_function
print(parameter_values['Negative electrode OCP [V]'])

#正极OCP
path, _ = os.path.split(os.path.abspath(__file__))
my_ocp_data_P = pybamm.parameters.process_1D_data("my_ocp_function_P.csv", path=path)
# Define custom OCP function
def my_ocp_function_P(sto):
    name, (x, y) = my_ocp_data_P
    #print(name,(x,y))
    return pybamm.Interpolant(x, y, sto, name=name, interpolator="cubic")
#parameter_values['Positive electrode OCP [V]'] = my_ocp_function_P
print(parameter_values['Positive electrode OCP [V]'])

#设置网格参数
var_pts = {
	"x_n":20,
	'x_s':20,
	"x_p":20,
	"r_n":30,
	"r_p":30
}

#试验步骤设置
exp = pybamm.Experiment(
[(
'Rest for 10 minutes',
'Discharge at 1C until 2.5V',
'Rest for 10 minutes',
'Charge at 1C until 4.2V',
'Hold at 4.2V until 0.05C')]*3,
	)

sim = pybamm.Simulation(model,  experiment = exp, parameter_values= parameter_values, var_pts = var_pts)
	#解析结果
sim.solve(calc_esoh=False)
	#sim.plot()
sol = sim.solution 
sol.save('1'+'.pkl')

sol3 = pybamm.load('1.pkl')
print(sol3.summary_variables.keys())

pybamm.plot_summary_variables(sol3)
```

### Relevant log output

```shell
KeyError: "'Capacity [A.h]' not found. Best matches are ['Measured capacity [A.h]', 'Throughput capacity [A.h]', 'Change in throughput capacity [A.h]']"
```

valentinsulzer, 2024-09-24:
Can you simplify this code more - removing everything not necessary to reproducing the error - so it's clearer what is happening? For example, it's not possible to test this locally since we don't have "my_ocp_function_P.csv". My guess is it's related to the saving and loading.

Yull-1, 2024-09-25:
> Can you simplify this code more - removing everything not necessary to reproducing the error - so it's clearer what is happening? For example, it's not possible to test this locally since we don't have "my_ocp_function_P.csv". My guess is it's related to the saving and loading.

Thank you for your response! I was able to resolve the issue by simply changing the line sim.solve(calc_esoh=False) to sim.solve(). This resolved the error related to 'Capacity [A.h]'.

However, I still have a question regarding the initial lithium concentration parameters. Specifically, how are the values for Initial concentration in positive electrode [mol.m-3] and Initial concentration in negative electrode [mol.m-3] typically determined? Are these values obtained through experimental methods, or is there a calculation method used to derive them? Any guidance on this would be greatly appreciated.

Thanks again for your help!

closed 2024-10-22 by kratman