pybamm-team/PyBaMM #4484: Remove `param = self.param` and use `self.param` directly instead

brosaplanella, 2024-10-03:
if these are the same, then why not just use self.param instead?

_Originally posted by @kratman in https://github.com/pybamm-team/PyBaMM/pull/4451#discussion_r1767514786_
            

There are 63 occurrences of `param = self.param` so I thought this would be better as separate issue.

brosaplanella, 2024-10-03:
I think @kratman also had some concerns about the performance of doing `param = self.param` instead of `self.param` directly, but please correct me if I am wrong.

agriyakhetarpal, 2024-10-03:
Using `param = self.param` creates a new reference (in terms of the immutability aspect) and lookup is faster, but we should prefer readability to performance. I would imagine that the performance impact would be more or less minimal and not actually measurable in a reliable manner.

We could switch to using `self.param` directly since it makes it clear we are working with an instance attribute

kratman, 2024-10-03:
For clarification what I was saying is that the benefits of a local reference (params) to the member (self.params) are
- shorter code by dropping the self
- potentially some time difference between accessing class memory vs function memory (never compared timings for this)

But the costs are
- hiding that the function is accessing class memory
- developers potentially needing to look up where the local reference is defined

I think it is better to use the member variable because being clear about which one is used is better for developers. In the background they both point to the same memory location since they are not copies but references.

brosaplanella, 2024-10-03:
With the comments above, a good compromise could be to keep the `param = self.param` in the basic models (which are aimed to users who might edit and build on) while for submodels we can switch to `self.param` as only developers will work on those.

kratman, 2024-10-03:
I would say that any memory benefits are minimal compared to the benefits of clarity. So using `self.param` is in general better. Everyone copies code around, so I don't think there is a benefit of saying that the base class must be one way and the submodels must be another way.

Trying to squeeze out performance by controlling access to member variables is basically the definition of pre-mature optimization. I would be more concerned with people not realizing that `param` was pointing to `self.param` in the class. Then they could modify `param` and not realize they were modifying `self.param` as well. This is why it is best to be clear by using `self.param`

PR #4494 (2024-10-10): Removes `param = self.param` to use `self.param` directly 
# Description

Used regex accompanied with some manual checking and have removed `param = self.param`  and have modified the code to use `self.param` directly.
Fixes #4484 

## Type of change

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that speeds up the code)
- [ ] Bug fix (non-breaking change which fixes an issue)

# Key checklist:

- [x] No style issues: `$ pre-commit run` (or `$ nox -s pre-commit`) (see [CONTRIBUTING.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CON [... cut]
closed 2024-10-10 by kratman