pybamm-team/PyBaMM #4488: Use `numpy.testing.assert_allclose` for more consistent floating-point comparisons

agriyakhetarpal, 2024-10-03:
Currently, we use `np.testing.assert_array_almost_equal` in 404 places across 41 files, but [the NumPy authors no longer recommend it](https://numpy.org/doc/stable/reference/generated/numpy.testing.assert_array_almost_equal.html#numpy.testing.assert_array_almost_equal). We should switch to the recommended [`np.testing.assert_allclose`](https://numpy.org/doc/stable/reference/generated/numpy.testing.assert_allclose.html#numpy.testing.assert_allclose) testing method where we can (which can be in one PR or multiple PRs), across all our tests.

Please note that there are a few differences in the number of arguments both functions take and in the default tolerance(s) set for them, so it's not a quick swap and some failures could be expected – but it should still be easy for someone to take it up in parts.

Saransh-cpp, 2024-10-04:
Also, all the instances of `assert all(numpy_array == numpy_array)`/`assert np.allclose(numpy_array == numpy_array)` (if we have any) should be replaced with `numpy.testing` calls.

mukulbindal, 2024-10-11:
Hi @Saransh-cpp @agriyakhetarpal If no one is working on this issue, kindly assign this to me.

arjxn-py, 2024-10-11:
Hi @mukulbindal, thanks for showing your interest. Feel free to pick this up and do let us know in case you need help :)

agriyakhetarpal, 2024-11-03:
Hi @mukulbindal, how is this coming along? Here's our guide in case you're not comfortable with contributing yet and if you need more instructions to follow along: https://docs.pybamm.org/en/latest/source/user_guide/contributing.html

Aswinr24, 2025-01-17:
Hi @agriyakhetarpal 
I am interested in working on this and would love to contribute by resolving it.
If it’s alright, I’d like to begin exploring and addressing this.
Thank you !

agriyakhetarpal, 2025-01-17:
Hi @Aswinr24, thanks! I can see that @mukulbindal has not responded or created a PR yet, and it has been a while. Please feel free to take it up.

closed 2026-01-22 by MarcBerliner